gnea/grbl #3: Buffer usage in status reports

chamnit, 2016-09-25:
@vlachoudis brought up a point that the status report buffer usage should state number of bytes and blocks free, rather than in use. Doing it this way, doesn't require knowing how big the buffer is. 

After asking around, it doesn't seem like anyone uses the buffer state and don't have a reason to. The one instance that I know its being used heavily is made obsolete by introducing the new jogging mode to v1.1.

I propose to disable buffer usage by default, make it available for re-enable via settings mask, and tweak it to say available, not used. This is generally only useful for debugging an interface and doesn't need to take up serial TX bandwidth unless there is a good reason. Does anyone?

gerritv, 2016-09-25:
Ok with me to make it optional, for me it was only used when people (the laser users mostly) complained about jerky motion to isolate the cause.

chamnit, 2016-09-27:
Last night's v1.1b push changed the buffer state to show bytes and blocks available, rather than in use. It has been disabled by default, but maybe enabled via the $10 report mask.

closed 2016-09-27 by chamnit