gnea/grbl #5: Feed rate appears to be ignored in $J

gerritv, 2016-09-26:
When issuing a series of $J commands the feed rate reported in the \[JOG: response goes to max, e.g. 500 default setting even though F50 is specified in the $J command.

String used is: $J=G91G20Y0.1 F50

Grbl version 1.1a

chamnit, 2016-09-26:
@gerritv : Looked at the code. Did find an unrelated bug with setting G93/94 correctly. But, didn't find a cause. It think it has to do with the report inches '$' setting. Maybe you're seeing the values being reported as mm and mm/min?

gerritv, 2016-09-26:
If it were Units related I would expect some relationship to 2.54 conversion factor. Instead I acceleration curve in the F: responses. 
E.g. if I change $110 to 5000, I see ramping of F:61,F:127,F:127,F:350,F:450, etc on the responses when using F10 in the $J command string for X+ and X- jogging.
If on the other hand I set $110=50 and then send $J's, I get F:50 back regardless of what the F value is in the $J string. And no ramping, always F:50 in the status response.

I will send you an updated version of GrblPanel that illustrates the issue

chamnit, 2016-09-26:
@gerritv : I'm confused. There should be an acceleration curve from the feed rate responses because status reports are providing the actual speed of the machine. which changes based on where it is in acceleration curve and if it's rate limited. When programming a $J command with a G20, the F10 is being interpreted as 10 inches/min = 254mm/min, rather than 10mm/min. The conversion factor is 25.4, not 2.54.

When I change $110=5000, I don't get above F:254 when programming $J=G91G20X1F10, which is correct. If $110=50, it shows F:50 because it's rate limited to 50mm/min.

gerritv, 2016-09-26:
argh! thanks for clarifying

closed 2016-09-26 by gerritv