gnea/grbl #8: Flood Override Toggles status reports don't seem to be working

winder, 2016-10-14:
The flood toggle doesn't seem to be working as expecting. After sending `0xA0` `T:F` is only included included in one override report. Is there a reason it would be automatically turned off?

```
[verbose] <Hold:0|MPos:51.396,9.608,-3.176|Bf:0,38|F:0.>
[verbose] <Hold:0|MPos:51.396,9.608,-3.176|Bf:0,38|F:0.>
>>> 0xa0
[verbose] <Hold:0|MPos:51.396,9.608,-3.176|Bf:0,38|F:0.|Ov:100,100,100|T:F>
[verbose] <Hold:0|MPos:51.396,9.608,-3.176|Bf:0,38|F:0.>
[verbose] <Hold:0|MPos:51.396,9.608,-3.176|Bf:0,38|F:0.>
[verbose] <Hold:0|MPos:51.396,9.608,-3.176|Bf:0,38|F:0.>
[verbose] <Hold:0|MPos:51.396,9.608,-3.176|Bf:0,38|F:0.>
>>> 0x9e
[verbose] <Hold:0|MPos:51.396,9.608,-3.176|Bf:0,38|F:0.|Ov:100,100,100|T:S>
[verbose] <Hold:0|MPos:51.396,9.608,-3.176|Bf:0,38|F:0.>
[verbose] <Hold:0|MPos:51.396,9.608,-3.176|Bf:0,38|F:0.>
```

gerritv, 2016-10-14:
@winder The setting toggles each time you send the command. The T:F response comes back each time so in essence you/we need to track the state and toggle it in the GUI as well. 
You/we can also check to see the status of the override by $G response, but you cant get this if Grbl is in Hold state.
I had suggested a change to this protocol to have Grbl send e.g. T:F on set, and T:f on reset but Sonny thought we (GUI devs) might be tired of last minute changes :-)

winder, 2016-10-14:
I suggested something similar (on set/unset) for the pin state. This ticket
was a little different though. In the example I was expecting 'T:SF'. The
spindle toggle is sent along with every Ov report until it is disabled but
the flood toggle is only sent once regardless of whether the toggle was
switched back.

chamnit, 2016-10-14:
@winder @gerritv : I'd like to get this right now, before everything is completely set in stone. There is a difference between how spindle stop and coolant toggles work. Spindle stop has a clear state whenever it's enabled or disabled, but coolant toggles don't. They will toggle what ever is active at the time, so a `T:F` message can mean EITHER it turned on or turned off. 

Gerrit's idea of showing upper and lower case to indicate which one is a good idea. However, coolant toggle also have another strange property (this feature was pulled from Haas' coolant toggle). They remain toggled until another toggle or a coolant command. So the question is then, when do you make it not appear in the status reports? Do you have to track when a change occurs?

winder, 2016-10-14:
@chamnit in that case, I would prefer that `T:F` is always in the status whenever M7 is enabled. Similarly with `T:M` whenever M8 is enabled. That way an absence of the command would mean these are disabled. A lowercase f/m for disabled would make this a little easier. The way it is now I can't really report the state of coolant since I don't send periodic `$G` or parse gcode while a job is being sent.

Also with these suggestions, the spindle toggle option would now behaves backwards compared to the coolants. So perhaps `T:S` is also reported whenever the spindle is on. Right now `T:S` means "The spindle is supposed to be on, but it is disabled.".

gerritv, 2016-10-14:
Presently you send a report when the state changes for Flood/Mist. My thought was that when it toggles to On state, send a T:F or T:M response. When it toggles off, send a T:f or T:m response. That way the GUI stays in sync with Grbl. And we don't need to see the T: in every OVR: message. The T: responses would follow whatever Grbl does, with M8 or M7 etc. 
But perhaps I am confused. If a flood coolant override is active, what happens if a M9 is sent? Does the override cancel as well?

winder, 2016-10-14:
@gerrit I think sending `T:fm` on a M9 would close the loop and allow the GUI to stay in sync without querying `$G` during a job.

chamnit, 2016-10-14:
Hmm, you guys make a good point. A GUI does need to know the running state of the spindle and coolant at any given time. Especially with the overrides now. Prior to overrides, you could indirectly know the state by the g-code. Not any more. 

Maybe instead of `T:` toggle, we should really make this the running state of the spindle CW/CCW and coolant(s) enable. It adds a few more bytes of data that needs to be sent, but I think it can be kept with the override data to reduce the communication overhead.

I'm warming up to the idea of just showing state. The toggle overrides would just change the run state, rather than show it executed.

As a side note, I also realized last night that real-time spindle speed is also missing. This can be altered by the spindle speed override. A GUI will have no idea what the current spindle speed is, at least not easily. I'll open another thread on this.

chamnit, 2016-10-18:
@winder @gerritv : Ok pushed an interface tweak that alters the `T:` toggle override report to show `A:` an accessory state. Character codes follow the `A:` to indicate if the spindle is enabled CW or CCW, and if flood/mist coolant is enabled. It reads the pin state directly to minimize any error.  

Any toggle override or g-code command that alters its state should trigger an immediate report update. Meaning it'll show up within 2 reports of it altering.

closed 2016-10-18 by chamnit

gerritv, 2016-10-18:
I will take a look at the change. I will make my override buttons toggle their colour to show the override active or not (and not rely on T: responses) , the A: info can then show the actual item state.