gnea/grbl #9: Spindle speed in status reports

chamnit, 2016-10-14:
Real-time spindle speed seems to have been forgotten in the status reports. It wasn't needed before, but with spindle speed overrides (and lasers), you'll have no idea what the spindle speed Grbl is executing at any particular moment. 

To address this, I'm proposing that we change the realtime "Feed Rate" report field to a "Feed and Speed" field. So from

`F:100.`

to 

`FS:100.,9000`, where `9000` is the spindle RPM with speed overrides.

gerritv, 2016-10-14:
Sounds good to me! I think F: is already the actual feed rate, including ramping etc. so adding actual Spindle speed fits well.

chamnit, 2016-10-18:
Real-time spindle speed is now included as part of a standard status report. Instead of `F:500`, you'll see an `FS:500,8000`, where 8000 is the real-time spindle speed in RPM. This is important because spindle speed and stop overrides can alter this state from programmed. GUIs need this feedback to know exactly what Grbl is outputting.

NOTE: `F:500` should still be supported, because this is what will show up if VARIABLE_SPINDLE is disabled in config.h. So check for both.

closed 2016-10-18 by chamnit