gnea/grbl #12: Spindle override on off error in GUI grbl panel

tklus, 2016-10-18:
When I pause the program and turn the spindle off via the spindle override button it gives me an error. The error reads "Fatal error on write to Grbl, the I/O operation has been aborted because of either a thread exit or an application request"

chamnit, 2016-10-18:
@gerritv : Any insight what may be causing this type of error in Grbl-Panel?

gerritv, 2016-10-18:
@tklus Please open an Issue on GrblPanel, this isn't a Grbl problem anymore. When you write it up can you clarify what buttons you are clicking and what is running to get to this condition? I tried to reproduce and couldn't.

chamnit, 2016-10-27:
@tklus : Spindle speed override should now be able to be changed while in a HOLD state. Could you download and test if this works? Also, if you can, please check if the spindle stop behavior is correct. Thanks!

tklus, 2016-10-27:
I just tried compiling and got this error.

## C:\Users\Tim.Klus\Documents\Arduino\libraries\grbl11c102716/grbl.h:101:4: error: #error "Override refresh must be greater than zero."

   #error "Override refresh must be greater than zero."

    ^

exit status 1
Error compiling for board Arduino/Genuino Uno.

chamnit, 2016-10-27:
@tklus : Make sure you delete any other existing grbl libraries. Just to be sure, I successfully compiled both via makefile and the Arduino example methods.

This compile error happens if you alter the REPORT_OVR_REFRESH_IDLE_COUNT value and if it's zero.

tklus, 2016-10-27:
I deleted all old GRBL libraries and I am still getting the error. any other ideas why? I am using the ide 1.6.9

chamnit, 2016-10-27:
Not sure. I just tested compiling on 1.6.12 and 1.6.7. Around v1.6.9, Arduino switched avr gcc versions. It may be buggy.

tklus, 2016-10-27:
i will try a new version of the ide. Weird thing is, I compiled 1.1c with no problems...

tklus, 2016-10-28:
I updated my Arduino ide to v1.6.12 and re downloaded the grbl files and tried compiling on a different computer even and I am still getting the errors. has anyone else besides Sonny tried compiling 1.1D?

[image]

chamnit, 2016-10-28:
@tklus : Try using the grblUpload example sketch. You shouldn't have to add all of the includes yourself. The order of them is very important.

tklus, 2016-10-28:
Well that was easy... it works fine now!

I was trying to add the grbl library like I had done in the past.

Thanks for the help!

cprezzi, 2016-10-30:
Is there a reason why only +/- chages are available for feed an speed overrides, but no direct set of a % value? It would be more convinient to have a slider for these values in the host software (instead of +/- buttons).

cprezzi, 2016-10-30:
Beside of that smal l detail - a BIG THANKS for your great work!

chamnit, 2016-10-31:
@cprezzi : Setting override values has been requested in the past, but it did not make it into the release for various reasons. The increment method lent to a very lightweight implementation that would fit into what's left in the 328p. I plan to address that in later versions. 

FWIW, the increment method is common on industrial machines since it's easier to just press a button to sneak up to an optimal feed rate or spindle speed. If you accidentally enter a bad value while the job is running, you can easily damage the machine, tools, and/or workpiece or put yourself in danger of being injured. It's arguable that the incremental method is safer since it's harder to put the machine into a situation like this accidentally.

cprezzi, 2016-10-31:
Ok, understand the security concern (which could also be handled in host software). It's not so important on laser cutters, as nothing can be damaged with qick power- or feedchanges (exept the workpiece).
Your solution has the nice option to directly connect +/- buttons to the controller, which I'd like to test onece.

Excellent work anyways!

closed 2016-11-04 by chamnit