gnea/grbl #22: How to setup inverting PWM signal and using +5v to switch off spindle motor?

viewsat, 2016-10-26:
Hi,
How to setup inverting PWM signal and using +5v to switch off spindle motor?

chamnit, 2016-10-26:
There is no way to do this currently. Do you have a specific need for this?

viewsat, 2016-10-26:
Because I bought the tb6560 drive its circuit design needs this, I have GRBL 0.9j modified, but Grbl 1.1 can not be used, this is casiobearing me to modify the completed program.
https://github.com/grbl/grbl/issues/918

viewsat, 2016-10-26:
##  GRBL 0.9j inverting PWM signal and using +5v to switch off spindle motor modified spindle control.h and spindle control.c these two files 

/*
  spindle_control.h -
*/ 

#ifndef spindle_control_h
#define spindle_control_h 
#define COMB_BIT0 COM2A0

// Initializes spindle pins and hardware PWM, if enabled.
void spindle_init();

// Sets spindle direction and spindle rpm via PWM, if enabled.
void spindle_run(uint8_t direction, float rpm);

void spindle_set_state(uint8_t state, float rpm);

// Kills spindle.
void spindle_stop();

#endif

----------------------------------------------------------------------------------------------
/*
  spindle_control.c -
*/ 
#include "grbl.h"

void spindle_init()
{    
  // Configure variable spindle PWM and enable pin, if requried. On the Uno, PWM and enable are
  // combined unless configured otherwise.
  #ifdef VARIABLE_SPINDLE
    SPINDLE_PWM_DDR |= (1<<SPINDLE_PWM_BIT); // Configure as PWM output pin.
    #if defined(CPU_MAP_ATMEGA2560) || defined(USE_SPINDLE_DIR_AS_ENABLE_PIN)
      SPINDLE_ENABLE_DDR |= (1<<SPINDLE_ENABLE_BIT); // Configure as output pin.
    #endif     
  // Configure no variable spindle and only enable pin.
  #else  
    SPINDLE_ENABLE_DDR |= (1<<SPINDLE_ENABLE_BIT); // Configure as output pin.
  #endif
  
  #ifndef USE_SPINDLE_DIR_AS_ENABLE_PIN
    SPINDLE_DIRECTION_DDR |= (1<<SPINDLE_DIRECTION_BIT); // Configure as output pin.
  #endif
  spindle_stop();
}

void spindle_stop()
{
  // On the Uno, spindle enable and PWM are shared. Other CPUs have seperate enable pin.
  #ifdef VARIABLE_SPINDLE
	SPINDLE_PWM_DDR &= ~(1 << SPINDLE_PWM_BIT); // Configure as normal output pin.

#ifdef INVERT_SPINDLE_ENABLE_PIN
	SPINDLE_PWM_PORT &= ~(1 << SPINDLE_ENABLE_BIT);
#else
	SPINDLE_PWM_PORT |= (1 << SPINDLE_ENABLE_BIT);
#endif

    #if defined(CPU_MAP_ATMEGA2560) || defined(USE_SPINDLE_DIR_AS_ENABLE_PIN)
      #ifdef INVERT_SPINDLE_ENABLE_PIN
		SPINDLE_ENABLE_PORT |= (1<<SPINDLE_ENABLE_BIT);  // Set pin to high
	  #else
		SPINDLE_ENABLE_PORT &= ~(1<<SPINDLE_ENABLE_BIT); // Set pin to low
	  #endif
    #endif
  #else
    #ifdef INVERT_SPINDLE_ENABLE_PIN
	  SPINDLE_ENABLE_PORT |= (1<<SPINDLE_ENABLE_BIT);  // Set pin to high
	#else
	  SPINDLE_ENABLE_PORT &= ~(1<<SPINDLE_ENABLE_BIT); // Set pin to low
	#endif
  #endif  
}

void spindle_set_state(uint8_t state, float rpm)
{
  // Halt or set spindle direction and rpm. 
  if (state == SPINDLE_DISABLE) {

    spindle_stop();

  } else {
	  SPINDLE_PWM_DDR |= (1 << SPINDLE_PWM_BIT); // Configure as PWM output pin.

    #ifndef USE_SPINDLE_DIR_AS_ENABLE_PIN
      if (state == SPINDLE_ENABLE_CW) {
        SPINDLE_DIRECTION_PORT &= ~(1<<SPINDLE_DIRECTION_BIT);
      } else {
        SPINDLE_DIRECTION_PORT |= (1<<SPINDLE_DIRECTION_BIT);
      }
    #endif

    #ifdef VARIABLE_SPINDLE
      // TODO: Install the optional capability for frequency-based output for servos.
      #ifdef CPU_MAP_ATMEGA2560
      	TCCRA_REGISTER = (1<<COMB_BIT) | (1<<WAVE1_REGISTER) | (1<<WAVE0_REGISTER);
        TCCRB_REGISTER = (TCCRB_REGISTER & 0b11111000) | 0x02 | (1<<WAVE2_REGISTER) | (1<<WAVE3_REGISTER); // set to 1/8 Prescaler
        OCR4A = 0xFFFF; // set the top 16bit value
        uint16_t current_pwm;
      #else
        TCCRA_REGISTER = (1<<COMB_BIT) | (1<<COMB_BIT0) | (1<<WAVE1_REGISTER) | (1<<WAVE0_REGISTER);
        TCCRB_REGISTER = (TCCRB_REGISTER & 0b11111000) | 0x05; // set to 1/8 Prescaler (原來0x02->7.8k, 修改0x05->487Hz,[2016-0827])
        uint8_t current_pwm;
      #endif

      #define SPINDLE_RPM_RANGE (SPINDLE_MAX_RPM-SPINDLE_MIN_RPM)
      if ( rpm < SPINDLE_MIN_RPM ) { rpm = 0; } 
      else { 
        rpm -= SPINDLE_MIN_RPM; 
        if ( rpm > SPINDLE_RPM_RANGE ) { rpm = SPINDLE_RPM_RANGE; } // Prevent integer overflow
      }
      current_pwm = floor( rpm*(PWM_MAX_VALUE/SPINDLE_RPM_RANGE) + 0.5);
      #ifdef MINIMUM_SPINDLE_PWM
        if (current_pwm < MINIMUM_SPINDLE_PWM) { current_pwm = MINIMUM_SPINDLE_PWM; }
      #endif
      OCR_REGISTER = current_pwm; // Set PWM pin output
    
      // On the Uno, spindle enable and PWM are shared, unless otherwise specified.
      #if defined(CPU_MAP_ATMEGA2560) || defined(USE_SPINDLE_DIR_AS_ENABLE_PIN) 
        #ifdef INVERT_SPINDLE_ENABLE_PIN
          SPINDLE_ENABLE_PORT &= ~(1<<SPINDLE_ENABLE_BIT);
        #else
          SPINDLE_ENABLE_PORT |= (1<<SPINDLE_ENABLE_BIT);
        #endif
      #endif
      
    #else   
      #ifdef INVERT_SPINDLE_ENABLE_PIN
		SPINDLE_ENABLE_PORT &= ~(1<<SPINDLE_ENABLE_BIT);
	  #else
		SPINDLE_ENABLE_PORT |= (1<<SPINDLE_ENABLE_BIT);
	  #endif
    #endif

  }
}

void spindle_run(uint8_t state, float rpm)
{
  if (sys.state == STATE_CHECK_MODE) { return; }
  protocol_buffer_synchronize(); // Empty planner buffer to ensure spindle is set when programmed.  
  spindle_set_state(state, rpm);
}

chamnit, 2016-10-26:
@viewsat : I'll add this to my to-do list, but it might be awhile. It's not a high-priority item. If you're successful in either updating the code yourself and convincing someone to help, please share the code. It'll help speed up the process to integrate it into the main repo here.

jahnj0584, 2016-10-26:
And the relay? Is it just one to spindle EN and the other the 5V?

dingorock, 2017-08-13:
@viewsat See Pull Request #246 [https://github.com/gnea/grbl/pull/246](url)
This simple change worked for me.

aqibi2000, 2017-11-04:
[image]

Very easy to invert the PWM signal in hardware at a cost of £0.10.

Make both resistors a value of 1K Ohm, use a NPN transistor e.g. 2N3904 and finaly ensure the V+ is connected to the 5V.

 You're welcome.

Apidde8, 2017-12-01:
Tks aqibi2000...

muxa, 2018-11-17:
@aqibi2000 thanks for providing schematics for the inverter. is this inverter discrete? or will it linearly invert 0-5V to 5-0V? E.g. if input is 1V the output would be 4V.