gnea/grbl #25: help/advice/maybe a simple update?

warwickben, 2016-10-28:
i'm new to git hub and mainly a chip maker that dabbles in electronics for fun, so sorry if this is the wrong place. if it is the wrong place any git users tell me where to re post this and i will.

i would like to try out the new beta but confused about how to up load it to my arduino.
i have a spare one i would like to install it to so i don't have to reconfigure my old one.
just pull the custom shield i made and swap the arduino's is what ill be doing.

do i just download the 1.1 zip like i did with the .9j and import the library in arduino - up load sketch?
do i need to need to delete the old library?

any way to change the current 1.1 if you do upload with the arduino ide that the libary/example name says some thing like grbl1.1 beta etc so i dont have to unstall the .9j example i have?

sorry if that sounds lazy.... 
i'm finishing up a cnc control panel like you see on a professional machine. so i dont need a laptop or pc near by.
basically a sealed box with a swing arm mounted to my mill.
10" touch screen, keyboard and track pad, buttons for different features etc. raspberry pi 3 , arduino with a custom shield for easy wiring.

jahnj0584, 2016-10-28:
Delete old library or rename the new one. The arduino IDE will get confused
with multiples and use the oldest file first - which would be grbl 0.9

warwickben, 2016-10-28:
> so i went to my library folder, changed .9 to say grbl9 then uploaded the new one and then changed that to say grblbeta.

thanks i wasn't sure if it was that easy, i was worried that even renaming them would cause problems with the sub files since i changed a folder name.

closed 2016-10-28 by chamnit