gnea/grbl #26: CoreXY homing wrong direction

chamnit, 2016-10-28:
https://github.com/grbl/grbl/issues/1107#issuecomment-256955403

Works on v0.9j, but not v1.1. Need to do a diff.

LCPINAGE, 2016-11-06:
@chamnit can I help here somehow?

chamnit, 2016-11-07:
@LCPINAGE : Yes. You can help by testing it when I issue the fix, but I'm pretty busy with work at the moment. Hoping to address all remaining issues later this week.

LCPINAGE, 2016-11-07:
Great, thanks!

chamnit, 2016-12-03:
@LCPINAGE : I'm having trouble finding anything wrong with the code, or rather any significant differences between the CoreXY in v0.9j and v1.1, that would cause the end of your homing cycle to move the other direction at the end of it. Do you have any startup lines active? Are you certain that your setting are identical between v0.9j and v1.1?

LCPINAGE, 2016-12-03:
@chamnit  working on it at this momment, will double check and let you know!

chamnit, 2016-12-03:
@LCPINAGE : I ended up bench testing this issue and was able to reproduce the problem. I found the bug and have fixed it. It'll be in the next push. Thanks for being patient.

closed 2016-12-03 by chamnit

LCPINAGE, 2016-12-03:
@chamnit great!!!!!!! Thank you for this awesome job overall!!!