gnea/grbl #27: Dsicussion, new feature: Spindle Coolant Temp monitoring with possible alarm

liqdfire, 2016-10-28:
It appears that there is a trend towards running micro cnc machines with water cooled cnc spindles. For water cooled spindles, having a good flow, and good cooling is critical to the life of the spindle. 

Having the ability for grbl to monitor a temperature probe that is inside the cooling loop could provide extremely valuable information as to the health of the system. Additionally, being able to trigger alarms based on thresholds for nominal coolant temp that could be used to shut the machine down, or put everything in sleep mode could save expensive repairs if something was to fail in the cooling loop.

Marlin 3D printer Firmware was orignally forked from grbl, and has temp probe monitoring. I would think that could serve as a strong base for implementation; though not positive the full array of temp sensors would need to be supported in grbl. 

This is the particular device I have in my cooling loop
https://www.amazon.com/gp/product/B00CMR3CC2

Thoughts?

chamnit, 2016-10-28:
@liqdfire : Grbl on the 328p is limited to what's remaining in flash and memory, which is virtually none. It is completely maxed out. However, there is a sister-project called Grbl-Mega that uses the much larger (flash and memory wise) chipset, which Marlin is built on. 

After Grbl v1.1 goes to master, I'll update the Grbl-Mega project to v1.1. From there, I'll lessen the development restrictions and let users help move that project forward. We will be able to add things like temperature monitoring there, along with other things like backlash comp. However, I don't personally plan on spending much time on Grbl-Mega. It's just there to bridge the gap until Grbl-ARM is released.

liqdfire, 2016-10-28:
@chamnit I was kind of afraid of that, though I had not dug in and compiled from the repo to get a sense of how much resources were available.

I know that the mega will have more than enough room, and I would be happy to contribute to the development of this feature for that platform.

Out of curiosity which platform or processor are you initialiy targeting with the arm version?

jahnj0584, 2016-10-29:
You could always get a 2nd arduino with a temp probe and have it set to
shoot an alarm whenever it reaches a certain temperature.

closed 2016-11-04 by chamnit