gnea/grbl #28: Spindle toggle behavior

winder, 2016-10-30:
I'm using the latest public changes on the edge branch: 
```
>>> $I
[VER:1.1d.20161027:]
[OPT:V]
```

While trying out the new accessory state status message and toggles I noticed some strange behavior with the `0x9e` spindle toggle. First of all, it doesn't seem to do anything (as per the documentation). Would it be possible to enable the toggle when idle?

I also noticed that when the machine is paused some `0x9e` disable commands are being ignored. The restore command always seems to work on the first try.

```
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,0>
>>> 0x9e
[MSG:Restoring spindle]
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000|Ov:100,100,100|A:S>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000|WCO:0.000,0.000,0.000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
>>> 0x9e   *** ignored ***
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
>>> 0x9e   *** ignored ***
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
>>> 0x9e   *** ignored ***
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000|Ov:100,100,100|A:S>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
>>> 0x9e
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,0|Ov:100,100,100>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,0>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,0>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,0>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,0>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,0>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,0>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,0>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,0|WCO:0.000,0.000,0.000>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,0>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,0>
>>> 0x9e
[MSG:Restoring spindle]
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000|Ov:100,100,100|A:S>
<Hold:0|MPos:25.644,9.608,-3.176|FS:0,1000>
```

The flood toggle works every time unless I send a bunch of them, so I don't think there's anything wrong with how I'm sending the command.

winder, 2016-10-30:
After a little more experimenting I can see that there is some sort of delay after re-enabling the spindle before I can toggle it back off. If that is a safety feature it seems to be backwards, if someone were to click the spindle toggle button repeatedly they could turn it off then back on very quickly and it would stay on. Putting a delay before you can toggle the spindle back on might be a good idea.

chamnit, 2016-10-30:
@winder : Good catch. There is an enforce delay to allow the spindle to get up to speed. It's an artifact of the automated parking routines. It doesn't make sense to do that for the overrides. I'll take them out in the next push.

closed 2016-10-30 by chamnit

jahnj0584, 2017-09-04:
@Channit, what is the spindle delay and can it be customized controller side?

chamnit, 2017-09-05:
It's only for parking motions.