gnea/grbl #29: $G reports M0

vlachoudis, 2016-11-01:
Is it normal while streaming and the status is "Run" when sending $G commands
that grbl reports M0?

chamnit, 2016-11-01:
@vlachoudis : No. Definitely not. It shouldn't show an M modal state for that. Looking back at older versions, it looks like this has been there for a really long time, all the way back to v0.8. Nice catch. 

I'll update v1.1 to not show anything unless M0/M1 pause or M2/M30 end are active. I'll also update the code to make sure they are shown correctly. M30 shows M2 in the g-code state. It'll be included in the next push. Hopefully tonight. Re-factored some laser code that needs a bit of testing first.

I'm not going to touch v0.9j, since it may break existing GUIs.

chamnit, 2016-11-04:
Fixed in this last push. M0 (or any program state) is no longer shown when in normal operation.

closed 2016-11-04 by chamnit

vlachoudis, 2016-11-04:
Thanks @chamnit