gnea/grbl #30: Spindle speed reporting wrong

109JB, 2016-11-01:
I noticed that when $13=1 to report inches, and then calling a $G, the S-value is wrong by a factor of 25.4. For example, if you Issue S500, the $G report shows S=19.7 if $13 is set to 1.

chamnit, 2016-11-01:
@109JB : I can't reproduce this issue. What version are you using?

109JB, 2016-11-02:
I'm not sure why but I can't reproduce now either..  My Arduino has been reset since then, but the firmware hasn't been updated since I saw this happen.  I am using Grbl 1.1c  [VER:1.1c.20161012:]

I know that it happened, because I thought it odd, and changing $13 would switch the reported S value between 500 and 19.7 and I realized it was a 25.4 factor. not sure what to say.

chamnit, 2016-11-04:
@109JB : I'm closing this issue until we can reproduce this error, but so far, it seems to be operating as intended.

closed 2016-11-04 by chamnit