gnea/grbl #31: Wow this is good! 1.1D and latest UGS Platform

splitn2, 2016-11-04:
Hi, well tonight I quickly flashed 1.1D into an existing Uno , a unit that started off in life running 0.9J then 1.1c and then tonight 1.1D. And also tonight loaded up the latest UGS Platform build. I am really impressed the overrides nd the visualiser are really really cool!!!!!! And so far although I have only been running test code its been running really well. Thanks @chamnit and @winder for your efforts and everyone else that contributes to these two great projects. This cnc is for a mate of mine who really wants to cut stuff and is not got enough time to be fluffing around with wiring etc so he gets me to do it, likewise I don't want him using something unreliable as it will inevitably soak up more of my time, so GRBL has been a good choice, simple and so far very effective. Cheers fellas!

chamnit, 2016-11-04:
@splitn2 : Thanks! Let us know if you find any oddities or bugs. I'd like to make sure everything is set before the master release. There have only been some minor implementation issues (oversights) and laser-mode requests, so I think I'll be close to ready to release it after this weekend. I'm going to spend a whole day trying to break it.

closed 2016-11-04 by chamnit

splitn2, 2016-11-04:
you are a clever and very generous man!