gnea/grbl #32: The PWM laser radotat incorrectly

AutoRich, 2016-11-12:
Hi!
Printed two files on GRBL v0.9j and GRBL v1.1d
There is a result:
[image]
The PWM laser it is displaced on one position to the left.
Correct please.
Test files:
[Test-S400.txt](https://github.com/gnea/grbl/files/587479/Test-S400.txt)
[Test-S400L.txt](https://github.com/gnea/grbl/files/587480/Test-S400L.txt)

chamnit, 2016-11-13:
@AutoRich : The PWM calculation was actually updated to be more correct, compared to the v0.9 version. So, I'm not too surprised that you are getting slightly different results. 

However, I did find a small bug with the altered calculation. If what you are seeing is related to that. Hang tight. Just note, that you may still see a slight difference between v0.9 and v1.1 once that fix is posted due to the previously mentioned calculation update.

chamnit, 2016-11-13:
@AutoRich : Should be fixed.

closed 2016-11-13 by chamnit