gnea/grbl #33: 1.1d (20161311) : Change unit and then send '?'

LETARTARE, 2016-11-13:
@chamnit
I congratulate you on your work to come up with Grbl-1.1.
Many improvements since version 0.7 (2012)!

Issue:
After the start-up and a first command '?' the report contains 'WCO' which can be stored in 'lastWCO'.
But after a change of unit, the first command '?' the report does not (usually) contain 'WCO'.
The calculation 'workPos = Mpos - lastWCO' becomes false because 'Mpos' uses one unit (eg mm) and 'lastWCO' uses another one (inches).
Does the GUI have to make the correction or Grbl could systematically send 'WCO' after a change of unit?

What is your opinion ?

******* Traduct ty Google *****************

chamnit, 2016-11-13:
@LETARTARE : Good catch. WCO should be updated immediately after a $13 unit change. I'll make sure I fix this in the next push.

chamnit, 2016-12-04:
Fixed in v1.1e

closed 2016-12-04 by chamnit