gnea/grbl #36: 1.1d (20161311) :  "$G"

LETARTARE, 2016-11-16:
in"[GC:] G-code Parser State Message"
you specify the value returned by Grbl when ordering "$G" 
[GC:G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]

but now I get after starting
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0] in mm
[GC:G0 G54 G17 G20 G90 G94 M5 M9 T0 F0.0 S0] in inches
'M0' is absent,
F0 is integer (mm) and float (inches).
The planner buffer is empty (no motion).

S0 is an integer or a float (S0.) ?
What little be the reason of the absence of 'M0' ?
Why differentiate 'Fxxx' ?

chamnit, 2016-11-16:
@LETARTARE : I need to document this. These were relatively recent tweaks. 

- M0 shown in the [GC:] state was incorrect. It has been since v0.8. M0 is program pause and was shown while in a run state. It shouldn't show anything. Hence, it's missing. It'll show when M0 is active, as well as, M2 or M30, if either of those are active.

- Floating point numbers without any precision digits are now shown as integers because it was redundant. If you alter the precision in config.h for spindle speed to something like 1, you'll see `S0.0`, rather than `S0`.

To answer your last question, displayed floating point precision is defined in config.h. In most applications, rate values in mm/min don't need a precision less than zero and in/min less than one. That's why there is a difference.

closed 2016-11-16 by chamnit

LETARTARE, 2016-11-16:
Thanks @chamnit