gnea/grbl #37: Single limit pin question

109JB, 2016-11-26:
I am putting together a new machine and would like to use a single I/O pin for all 3 limits.  I believe the following modifications will make this work, but am unsure since I am not that good a programmer. I intend to use digital pin #9 as the I/O pin for all limits. Can someone check my changes below to see if these are all that I need to do?

In cpu_map.h, I changed the following lines with the changes in Bold italic text (Comments omitted)
Line 63: #define X_LIMIT_BIT      1 
Line 64: #define Y_LIMIT_BIT      _**1**_  
Line 64: #ifdef VARIABLE_SPINDLE 
Line 65:     #define Z_LIMIT_BIT     _**1**_ 
Line 66:   #else
Line 67:     #define Z_LIMIT_BIT     _**1**_  
Line 68:   #endif

In config.h I set the homing bitmasks as follows:
#define HOMING_CYCLE_0 (1<<Z_AXIS)  
#define HOMING_CYCLE_1 (1<<X_AXIS)
#define HOMING_CYCLE_2 (1<<Y_AXIS)

I'm thinking these are the only changes I would need and would result in completely homing Z first then X then Y.  Have I got this right?

jahnj0584, 2016-11-26:
you would generally home Z, Y, and X alone in that order. X/Y doesnt really
matter, as long as your Z is raised up.

109JB, 2016-11-26:
That wasn't the question. The question is if I changed the code appropriately to use a single I/O pin for all 3 limits.

chamnit, 2016-11-26:
@109JB : I can't say for sure but I think that is about all you have to do. The homing cycle just polls the pins specified. The only thing that would break is the pin state reporting. You may want to alter that to show XYZ anytime the pin is triggered in report.c. There would also be some redundant code in the hard limit handler but it should work fine.

chamnit, 2016-11-26:
@109JB : If you test this and it works, I'll add this as a build option.

109JB, 2016-11-26:
Ok. I'll test it and report back. I have to install the switches, and have family stuff going on today and tomorrow, but should be able to get them installed on Monday.

As an aside, the reason for this is because I am using sourcing npn proximity sensors on 36v that would fry that I/O pin, so will wire the prox sensors to a single voltage divider circuit to drop the pin voltage to 5V.

109JB, 2016-11-29:
I finally got everything I needed to check single pin limits/homing set up and am happy to report that the changes I listed in post #1 in this thread worked. Those were the only code changes I made and I ran about 30 homing cycles without a hitch.

One thing to note though. When I used the Arduino IDE to upload I got errors using the "AVRISP MKII" programmer. I set this to the "AVRISP" programmer and it uploaded fine. This was on a Nano board just for testing. I will switch to an UNO before cutting chips, but thought I'd put this out there.

chamnit, 2016-11-30:
@109JB : Great to hear. Thanks for testing. I'll make some of the necessary mods for everything else to work with this.

109JB, 2016-11-30:
@chamnit No problem.  I can't do much in the way of coding to help you, but am happy to do what I can to support the cause.

I also did some more testing of the upload error that I encountered.  Not sure what is going on but it seems to be limited to that one Nano board I am using .Other Nano boards upload fine with the "AVRISP MKII" programmer.  However, I do get a few warnings when uploading. The warnings I get relate to redefinition of constants. GRBL works fine though.

trbLeeciN, 2017-12-15:
Hi there, this single limit pin may cause problems. 
For example, you start homing while limX already triggered, it will think that limit Z is triggered. So homing will fail. To do single limit pin homing you need to make sure that none of the limit switches are triggered.

109JB, 2017-12-15:
That is correct, but I have been using it for just over a year now without that ever happening. I have my machine set up for homing only with soft limits activated and hard limits not activated (IE: $20=1, $21=0, $22=1) Even if one is triggered before starting homing, all that I can see happening is the machine won't home.

lhartmann, 2018-03-08:
This same concept could also incorporate the probe pin, could it not? Also a simple boot check could throw grbl into alarm state if a shared limit pin is initially triggered. The user would have to manually clear the alarm, move the machine to a safe starting position, and reset.

HorstBaerbel, 2018-04-14:
I have a similar problem here and hope you have advice for me. I have a machine with all limit switches serially connected to each other and switching to ground. Switches are NC / normally closed, so I have set up:  
#define DEFAULT_INVERT_LIMIT_PINS 1
and connected GRBLs XYZ limit input pins to the single switch line on the machine. Now the machine does move only if I activate one of the switches as if DEFAULT_INVERT_LIMIT_PINS does not work. Am I missing something?