gnea/grbl #38: Laser mode issues

HansPeterHaastrup, 2016-11-27:
Hi there,
I'm running the 1.1 beta on a Raspberry Pi CNC Shield from Protoneer. I'm using this to control a homemade 40W CO2 laser combined with LaserWeb3.
I switched fron 0.9j because of the laser mode - which looks promising!

My findings so far:

- I miss the comments on the settings when doing a '$$' command. It might be a compile flag that I'm missing?

- I'm compiling on the pi that is connected to the shield. Verbose programming shows some warnings about true, false, max and min being declared multiple times. it's not a huge thing but warnings nonetheless.

- The system stops at an arbitrary point and gives an 'Alarm' state - Is there any way to get detailed information about what caused the alarm?

cprezzi, 2016-11-27:
You can find infos about config options, error messages and alarm states in https://github.com/gnea/grbl/blob/edge/doc/markdown/interface.md.

HansPeterHaastrup, 2016-11-27:
It looks more like the whole system just resets and then (by default) starts up in a locked state - thus the alarm

Here's the terminal output (from LaserWeb3)
Sent: G1 X93.000 S70 Q: 59019
Recv: ok
Sent: G1 X92.900 S70 Q: 59018
Recv: ok
Sent: G1 X92.800 S70 Q: 59017
Recv:
Recv: Grbl 1.1d ['$' for help]
Recv: [MSG:'$H'|'$X' to unlock]
Recv: <Alarm|WPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
Recv: <Alarm|WPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
Recv: <Alarm|WPos:0.000,0.000,0.000|FS:0,0>
Recv: <Alarm|WPos:0.000,0.000,0.000|FS:0,0>

And here's the gcode where it stops:
G1 X93.100 S70
G1 X93.000 S70
G1 X92.900 S70
G1 X92.800 S70
G1 X92.700 S70
G1 X92.600 S70
G1 X92.500 S70
G1 X92.400 S70
G1 X92.300 S70
G1 X92.200 S70
G1 X92.100 S70
G1 X92.000 S70

Nothing unusual I guess.

HansPeterHaastrup, 2016-11-27:
These are my settings:
: $0=10
: $1=25
: $2=0
: $3=3
: $4=0
: $5=0
: $6=0
: $10=0
: $11=0.010
: $12=0.002
: $13=0
: $20=0
: $21=0
: $22=1
: $23=3
: $24=500.000
: $25=1000.000
: $26=250
: $27=2.000
: $30=1000
: $31=0
: $32=1
: $100=80.000
: $101=80.000
: $102=5120.000
: $110=1000.000
: $111=1000.000
: $112=200.000
: $120=500.000
: $121=500.000
: $122=10.000
: $130=550.000
: $131=250.000
: $132=90.000

cprezzi, 2016-11-27:
Did you click the stop button at some time? 
You get the Grbl version string after a few lines, which is sent by grbl after a new connect or reset.

cprezzi, 2016-11-27:
By the way, your settings look fine.
Which exact version of grbl do you have installed? Should be the commit from 11/13/2016 or later.

chamnit, 2016-11-27:
@HansPeterHaastrup : 

- $$ settings strings are gone due to flash constraints. They take up about 0.5-0.8KB depending on how minimal you go. There is no longer space for this.

- I do not get warnings about multiple definitions. Please make sure any other existing Grbl libraries are removed from the Arduino paths. If you aren't using the Arduino IDE to compile, please explain what you are using and how you are compiling.

- Your alarm looks unusual. It has no indication of an alarm triggered. It seems like your Arduino did a HARD reset for some reason. I say this because it initialized with a [0,0,0] position, which only happens upon a power cycle. Something is causing your Arduino to hard reset, and it's hard to say if it's Grbl or something else. Can you disconnect everything from your Arduino and run your g-code program by itself? Does it do the same thing? If so, try another g-code streamer like UGS to eliminate the possibility that it's the GUI.

- The documentation has moved to the /doc folder, but will be up on the Wiki when v1.1 goes master.

Finally please paste your `$I` Grbl build info output. It will tell us what version you are using and what options you have enabled.

HansPeterHaastrup, 2016-11-27:
- $$ string gone: Ah, makes sense. It was a nice feature bue it makes sense that they had to go.
- Compiling: I'm using Arduino. I tried to get the edge from git and to follow the build instructions there. I couldn't import the folder as an Arduino library for some reason and ended up just opening and comiling the .ino file in the example folder (grblUpload). I'm usually on a windows machine so i might have done something wrong.
- Alarm: Yes, I figured after I wrote the last comment that it was a hard reset because I lost the machine position. I know it works fine if I'm not connected to the laser PSU so there is definetely some EMI involved in this. 
- Documentation: Thanks for the info!

$I: 
[VER:1.1d.20161112:]
[OPT:V] 

Right now I'm seeing an issue where when I issue $H all communication stops. No motor movement is happening. Only a hard reset will help here. When I then Reset, the laser tube is triggered in 3 short bursts :/

chamnit, 2016-11-27:
If it's EMI causing your hard reset, then this is not a Grbl issue.

A hard reset puts the Arduino pins in a floating state. There's not much Grbl can do about that either. I recall that the Arduino bootloader keeps the startup sequence held for a second or two. Flashing with a programmer via ISCP header gets rid of the bootloader. It can reduce or eliminate this boot sequence. I would try that if you have a programmer or a spare Arduino to use as a programmer.

closed 2016-11-27 by chamnit

Protoneer, 2016-11-27:
@chamnit would you agree that an EEPROM reset is needed after updating to v1.1?

(Done with "$RST=*" command)

chamnit, 2016-11-27:
@Protoneer : Yes, but Grbl should automatically do it upon the first power-up with the v1.1 firmware. It checks the settings version number (different from firmware version). v0.9j -> 9 and v1.1 -> 10.

cprezzi, 2016-11-27:
The behaviour of short laser fire is also happening when just reconnecting serial connection over usb (by software, cable stays connected), but this should not cause a reboot, right?

chamnit, 2016-11-27:
@cprezzi : Reconnecting over serial-USB will also hard reset the Arduino. That's the way its designed. See Arduino [docs](http://playground.arduino.cc/Main/DisablingAutoResetOnSerialConnection).

cprezzi, 2016-11-29:
Thank you, found it in the meantime ;)

Did you had time to look at why the gcode queue is not cleared after ctrl-x?
Or is there another way to stop the machine immediately and then dropping the queue?

I was thinking of disconnecting and reconnectig serial, but this will shoot the laser for short time, which is not good (safe).

chamnit, 2016-11-29:
@cprezzi : Sorry I haven't had a lot of time the past couple of weeks. Several reports are coming due and got a little sidetracked with researching Grbl ARM too. I try to look at it tonight. 

Yes, the uncontrolled firing of the laser upon reconnect is very unsafe, but I'm not entirely sure what to do about that. I would seriously try to write Grbl through the ISCP header with another Arduino or programmer. This will get rid of the boot loader and will likely greatly reduce or eliminate this problem. Unfortunately I can't test that theory right now.

cprezzi, 2016-11-29:
It's probably not very practical for normal users to program their Arduino though ICSP. I hope to find another solution, eventually hardware based. I think the problem is, that pin11 is floating during reset until it is configured as open collector output and pwm set to zero. If I could design a simple external circuit which pulls the pin to ground until the Arduino is ready, that could solve the problem.

I hope you can find a solution for the buffer dumping. An additional real time command to clear the buffer could help, so we could first hold and then reset the queue.

electrokean, 2016-11-29:
Have you tried just adding a high value pull down resistor?  Maybe start with 100k.

cprezzi, 2016-11-29:
Shure I will try a pull-down first ;)

chamnit, 2016-11-29:
@cprezzi : I'm not worried about the hold issue. It's likely just a state machine oversight on my part. A few lines of code should fix it. 

As far as an additional real-time command to clear a queue, I'm not going to do that. This will introduce a large set of runtime and safety conditions that will need to be accounted for and handled appropriately. This type of control is also often grossly misused when a controller allows it. Grbl intentionally restricts clearing the buffer with a Ctrl-X soft reset or power-cycle only. Primarily because, if you are in a situation that requires this, then you will very likely need to abort and restart your job.

If you are asking about better jogging handling, Grbl v1.1 has a new jog mode. It has a built-in way to clear the buffer when dealing with jogging motions. Only during the JOG state, a feed hold will bring the machine to a stop and then automatically clear the planner buffer. There is also a separate jog cancel real-time command that will perform the same task. It's highly recommended that you use this instead of the feed hold command, because it won't invoke a HOLD state accidentally after the jog comes to a stop.

cprezzi, 2016-11-29:
I was not talking about jogging, I'm just working on the pause and stop behaviours, which should stop immediately. For pause we send hold, which does stop immediately, but ctrl-x semms to play the rest of the queue before stopping. This problem just came up because we started using the character counting method and fill the gcode buffer in grbl. I'm not shure if my ctrl-x is even executed.

chamnit, 2016-11-29:
@cprezzi : I happened to have an Arduino with me and just did a quick test. Hold followed by a Ctrl-X works as it should. It'll abort the hold, reset Grbl, and flush the planner and serial buffers. 

I think I know where the problem is. Your streamer has to stop streaming g-code to Grbl prior to sending the Ctrl-X. Otherwise, Grbl will execute any g-code streamed to it after the reset completes. This occurs after a HOLD->reset, because Grbl re-initializes to the IDLE state. Not the ALARM state that blocks all g-code commands.

A HOLD->reset is a considered a "safe" reset, where position is still guaranteed by Grbl, so it doesn't have to invoke an alarm. It allows you to immediately continue operating the machine. If the machine were moving during a reset, this invokes an ALARM state because it has to halt the machine immediately beyond the acceleration settings and can't guarantee position. I hope that clears things up.

cprezzi, 2016-11-29:
I am stopping the streamer before sending ctrl-x but I think my ctrl-x (0x18) is not correctly sent, because i don't see any reaction from grbl. I still get response to status queries. If ctrl-x would be executed by grbl, I should get the welcome string "Grbl 1.1d ['$' for help]" right?

chamnit, 2016-11-29:
@cprezzi : Yes. You should see the welcome string after every reset. Grbl picks off real-time commands from the serial ISR directly. It never enters the serial RX buffer, where you are storing additional data with the CC protocol. So, as long as the Ctrl-X was sent, Grbl should have flagged the system and started the reset. My guess is that the Ctrl-X isn't being emitted for some reason.

cprezzi, 2016-11-29:
Ok, found the problem! I had to do a String.fromCharCode(0x18) to get node serial happy. 
Thank you for your patience and sorry for the confusion!

Protoneer, 2016-11-29:
On the floating pin topic...

What if we use a hardware AND gate that connects to Spindle Pin and Coolant/Spindle Direction pin to enable the laser?

With this configuration the Coolant/Spindle Direction pins needs to be high as well as the Spindle Pin to fire the laser.

electrokean, 2016-11-30:
You'd probably still need pulldowns on both outputs (AND gate inputs) as they'll be floating at reset and could still randomly trigger an active output.
The pulldown on the PWM output is probably sufficient in most cases.  You'd just need to confirm that the resistor doesn't have an impact on the PWM signal (dependant on laser drive input), which is why I recommended a high value of 100k.

atsage, 2019-02-26:
Did anyone fix the laser turning on?  Removing the bootloader did not fix the problem for me.