gnea/grbl #39: Problem with Pause->Stop behaviour

cprezzi, 2016-11-27:
I am working on improvements of LaserWeb3 especially for the new Grbl features and have some trouble with hold and reset (in combination with character counting protocol):

- For pause, we use '!' hold command to get immediate response, which works well with new grbl (but will be a problem if used with older grbl, as the laser will not be switched of!).
- Resume with '~' is also ok. 
- On stop we send '0x18' (ctrl-x) which works well while the job is running, but if the job was paused before, grbl remains in hold state an doesn't clear the queue.

@chamnit : Do you have any suggestion how to solve this issue?

chamnit, 2016-11-27:
@cprezzi :

- Not sure how I can help with people using older versions of Grbl with LW3. Most GUI devs check the welcome string or build info string for the version and support different protocols based on that. That said, Grbl doesn't officially support any laser mode prior to v1.1. The existing v0.9 laser firmwares are hacked forks. Hopefully, with v1.1 going to master, you'll just need to support v1.1 for LW3.

- ok.

- I'll check this behavior to make sure there isn't some bug in the v1.1 code. It's quite possible that it is one.

cprezzi, 2016-11-27:
We decided to drop support for grbl before v1.1 with an error message and link to this repo. I will change the link once you go master. Hope this is ok for you. 
I will check issues on this repo and try to help with problems related to LW3.

chamnit, 2016-11-30:
@cprezzi : Since you've noted that this problem is fixed [here](https://github.com/gnea/grbl/issues/38), I'm closing this issue.

closed 2016-11-30 by chamnit