gnea/grbl #42: Arduino Uno vs Nano

mayhem2408, 2016-11-30:
I can compile grbl 1.1 for the uno, but have some of the laser control boards with stepper drivers on them that use the nano, It will not compile because there is not enough flash memory available. Is there anything I can disable that will not be needed on a laser setup that will reduce the memory usage?

`Sketch uses 31,144 bytes (101%) of program storage space. Maximum is 30,720 bytes.`

chamnit, 2016-11-30:
@mayhem2408 : Are you enabling anything in the config.h that is not default?

mayhem2408, 2016-11-30:
CoreXY is the only change I have made. I tried disabling it but it still wont fit. The compiled size did not change.

chamnit, 2016-11-30:
@mayhem2408 : Make sure you edit the config.h in the Arduino library folder and all other Grbl libraries are removed. 

Enabling CoreXY will cause Grbl to compile larger than the 30.5KB flash limit of FTDI-based Arduinos, like the Nano. A default v1.1 build will be about 400KB under this limit and should work fine. 

If you can write a boot loader onto your Nano, try writing the Uno's on there.  It's pretty easy and just requires a spare Arduino (programmed as an ISP), setting the target board as an UNO (not Nano), and four wires. Afterwards, your Nano will be accepted as an Uno board and have 1KB of additional flash.

closed 2016-11-30 by chamnit

mayhem2408, 2016-11-30:
I had thought about changing the bootloader. I'll to it now an see how it goes.

mayhem2408, 2016-11-30:
I should have just done that earlier. Now I can upload it using the Uno setting the the Arduino IDE. No Errors and uploads are now quicker. Next is the try it out. Thanks for the tip.

`avrdude: reading on-chip flash data:`

`Reading | ################################################## | 100% 3.67s`

`avrdude: verifying ...`
`avrdude: 31144 bytes of flash verified`

`avrdude done.  Thank you.`