gnea/grbl #45: Settings Description ($$) ?

nsiatras, 2016-12-02:
I just uploaded the 1.1 to an Uno board and I am doing the necessary changes to my GCode sender. I noticed that the '$$' returns the list of the GRBL settings but **the description is missing**.

Is this a bug or the new version will not send the description ?

panos78gr, 2016-12-02:
It's gone!: 

"In prior versions of Grbl, the $ settings included a short description of the setting immediately after the value. However, due to flash restrictions, most human-readable strings were removed to free up flash for the new override features in Grbl v1.1. In short, it was these strings or overrides, and overrides won. Keep in mind that once these values are set, they usually don't change, and GUIs will likely provide the assistance of translating these codes for users."

gerritv, 2016-12-02:
You need to read the documents describing the new protocol. They are in the Doc directory.

nsiatras, 2016-12-02:
I know. I just implement the new thing with the Work Coordinate Offset in my gcode sender. At the moment I am implementing the new jogging features

closed 2016-12-02 by chamnit

winder, 2016-12-02:
Keep in mind that the description CSV files are in a machine readable format, so you can drop them in a resources directory and write a loader for easily keeping your settings up to date. The filename also has the locale in it, hopefully we'll get localized strings eventually as well.