gnea/grbl #46: Step by Step acceleration in a CPU-friendly way

jsb1, 2016-12-02:
@chamnit :Hello,
There is an interesting paper on the net:
[http://www.embedded.com/design/prototyping-and-development/4008259/Linear-motor-control-without-the-math](http://www.embedded.com/design/prototyping-and-development/4008259/Linear-motor-control-without-the-mathl)
I could implement it a way which allows running on 8-bit cpus. Really amazing: No float calculations, not division, no multiplication and no table lookups required in stepper interrupt. Just some adding and shifting - all 16bit.
Biggest advantage is saving a lot of cpu time: There are only 3 line segments needed: accelerate, cruise and deccelerate.
And some disadvantage: stepper interrupt has to run in fixed intervals and not at too high frequency (8-16 khz). Not a big problem: see comments in code.

Now I want to try it in grbl. But I don't want to break too much existing functionality in stepper.c.
Could You take a look at my code and give me some hints how to put that in? 
I think I can do the ISR. But how to calculate the line segments?
You can simply compile main.c on PC and play with it.

[steptimer.zip](https://github.com/gnea/grbl/files/627957/steptimer.zip)

chamnit, 2016-12-02:
@jsb1 : You're not the first person to mention this stepper algorithm. I've analyzed it several years back around the time I first started working on Grbl in 2011. In short, it's not as good or simple as it seems. There are a lot of problems with it, particularly at high step rates. I'll leave it up to you to figure out what I mean by this.

closed 2016-12-02 by chamnit

jsb1, 2016-12-02:
@chamnit :This is the only clue I have: interference between timer frequency and step frequency can easily hit mechanical resonances. Is it that?

usbcnc, 2016-12-02:
@jsb1, can you tell the difference on your machine if per step is implemented? I have done similar thing to both 0.9j and 1.0d. If you want I can send you a hex file to test it.

usbcnc, 2016-12-03:
@jsb1 @chamnit Here is the build for 0.9j with per step acceleration. I initially did it on STM32 and latter port back to Arduino. I do not have a good CNC machine to test. The ISR to handle the step pulse is shorter so I believe it allows fast step frequency.

[grbl0.9jwithperstepacceleration.zip](https://github.com/gnea/grbl/files/628653/grbl0.9jwithperstepacceleration.zip)

jsb1, 2016-12-06:
@usbcnc sorry, but there seem to be good reasons why ranade algorithm is not there anymore. And I will not try binaries :-).
@chamnit Yes I see, it was in 2012 and "thrown out" 2013. And I understand why. I suppose it works well with timer frequency much higher than maximal stepper frequency (nyquist?). So not on AVR... And with relatively hight computational cost on typical 32-bit controllers.

usbcnc, 2016-12-07:
@jsb1 I hear what you are saying. I will trying to add per step along with the ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING. However not sure if it will fit in 328p. Will work on ARM first. Also ordered a Mega2560 to experiment the theory.