gnea/grbl #49: min/max PWM variables and config.h

arkypita, 2016-12-04:
In version 1.1e I can find $30 and $31 that control max and min spindle speed, but in config.h exist macro SPINDLE_MINIMUM_PWM_VALUE.

This macro does something different from $31? Is it deprecated?

chamnit, 2016-12-04:
Yes. SPINDLE_MINIMUM_PWM_VALUE does something different than $31 min rpm. This sets the PWM range that the rpm range spans over. There should be an explanation right next to this option in the config.h file.

closed 2016-12-04 by chamnit

arkypita, 2016-12-19:
Hello Chamnit, excuse me if I reopen the discussion.
I have tried searching the macro SPINDLE_MINIMUM_PWM_VALUE in the code (gnea-master), and I found that it is only defined in config.h and not used in any part of the code. I guess it's functionality been lost with the rewriting of some parts.

I think it is a useful feature, because if I understand correctly, you can set a minimum PWM value to be used at the value $ 31.

In my case for example the laser in my possession does not turn on at all, if the duty cycle is lower than 11% (30/255), for which I would be useful to be able to readjust the 0-1000 value that imposed on $31 and $32 to a scale of values 30-255.

I hope to be re-integrated the functionality of SPINDLE_MINIMUM_PWM_VALUE in a future version of the code

regards

chamnit, 2016-12-19:
@arkypita : It's there. The macro is SPINDLE_PWM_MIN_VALUE.

arkypita, 2016-12-19:
Maybe it's only a "typo"

in config.h is defined as SPINDLE_MINIMUM_PWM**_VALUE** 

> // #define SPINDLE_MINIMUM_PWM_VALUE 5 // Default disabled. Uncomment to enable. Must be greater than zero. Integer (1-255).

but in cpu_map.h and in grbl.h

`  #define SPINDLE_PWM_MAX_VALUE     255 // Don't change. 328p fast PWM mode fixes top value as 255.
  #ifdef SPINDLE_MINIMUM_PWM
    #define SPINDLE_PWM_MIN_VALUE   SPINDLE_MINIMUM_PWM
  #else
    #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
  #endif`

`#if defined(SPINDLE_MINIMUM_PWM)
  #if !(SPINDLE_MINIMUM_PWM > 0)
    #error "SPINDLE_MINIMUM_PWM must be greater than zero."
  #endif
#endif`

chamnit, 2016-12-19:
@arkypita : You are certainly right. Fixed now. Weird thing is that I remembered altering those when I changed the names to be more consistent. Must have lost some work at some point.