gnea/grbl #54: Question about G4

arkypita, 2016-12-06:
When executing G4 command, real-time status report "Idle" even if pause command is not completed, and although there are other commands already queued for execution.

Some GUI (link mine) use tu push continuous "?" status request to know current position and to put the interface in the right operational mode (i.e. enabling homing button only when status is alarm/idle).
In this case UI is carried to think to be in Idle state and behaves in an unexpected way, enabling buttons that should not be enabled during the execution of a program.

I can not assert if this behaviour is an issue or not, but Idle is documented as:
> Idle: All systems are go, no motions queued, and it's ready for anything.

Here an example: https://youtu.be/OvhySa2IGdc

chamnit, 2016-12-06:
@arkypita : Yeah. This is an outstanding issue and an oversight by me. I did have a note to make G4 dwells show 'Run' rather than 'Idle', but forgot to install it. If I can easily I will, but it will not likely be fixed until v1.2.

yaroslavVl, 2018-12-13:
Problem with ENABLE signal in this case. During the dwell, ENABLE signal turns off the motors.