gnea/grbl #55: Setting to alter laser PWM frequency

chamnit, 2016-12-08:
From https://github.com/gnea/grbl/issues/34#issuecomment-262819761.

Add a '$' setting to allow different laser PWM frequencies. Will likely implement this as value with the current `$32` laser mode setting, such that `1`, `2`, or `3` would indicate different frequencies.

Need to research how PWM frequencies effect lasers and whether or not this is even required.

mayhem2408, 2016-12-08:
I run a 5W Laser diode and my TTL laser driver did not like the default 1KHz PWM. I changed mine to 8KHz and it runs perfectly. The drivers documented frequency is 10KHz, but 8 is close enough.

I think part of the problem was the speed I am running it. If I feed at F4800 and my resolution of the raster is 10 pixels per mm, that is 800 pixels per second. This probably means that some pixels only get 1 pulse, and some get 2. At 8KHz, each pixel get 10 shorter pulses and the result is much smoother.

So to answer your question, in my situation the PWM frequency does make a difference.

cprezzi, 2016-12-19:
I have a 5W laser engraver with a proprietary controller and software, that can change the pwm frequency (via cycle duration in ms). This gives me the option to optimize for good grayscale or depth map.

mayhem2408, 2016-12-19:
The PWM frequency can be changed in the cpu_map.h around line 140.

mayhem2408, 2016-12-19:
I have a question on the PWM timer. It appears to be an 8 bit timer which is not high enough for my next project.
I plan to upgrade my 50W Chinese CO2 laser control board to a GRBL 1.1 controller. The current control has a resolution of 0.1% or 50mW steps. I run it between 5-15% getting 100 steps of grey. Using GRBL PWM at 8 bit means about 200mW per step and only 25 steps of grey between 5-15%. 

I have a 5W laser which I use between 15-60% which gives me over 100 shades of grey and it looks fantastic.

Any thoughts on what could be done to improve the resolution when using a small power range.

chamnit, 2016-12-19:
@mayhem2408 : Hmm. Not 100% understanding what you are asking, but I think you are saying that your laser driver works optimally between a small range of control voltages. You'd like to have a higher resolution within this range. 

I'll be the first to admit that I'm not an expert at laser and their controllers. So, I'm not certain if there is another way to improve raster engravings, like altering the PWM frequency and feed rate.

For Grbl, there isn't another timer available with a higher resolution than 8-bit. The one 16-bit timer is being used by the stepper interrupts. So, I can't help you there. However, if the voltage range is low, like 1V-2V, you could try placing a voltage divider circuit on the PWM output. This will effectively scale the output down and should improve resolution by the voltage ratio. If you want a larger range, I think there might be a way to do it if you add a voltage level shifter to bring up the low side, in combo with a voltage divider. This is all theoretical though.

Another option would be to use a separate controller to convert Grbl's output to something your laser likes better. This could give you the best control, but it'll require some interrupt programming to ensure that the timing is still exact.

mayhem2408, 2016-12-19:
Not quiet. My 50W laser is extremely powerful. At 100% power it cuts 6mm plywood. If I want to do light engraving, I need to use very low power settings. The laser controller itself is TTL triggered, so it only accepts 0 and 5V signals. Sending it 4V is still one and 1V is still off. Using PWM at 8KHz+ allows power modulation.

I have an idea of using another Arduino to simple control the PWM at a higher resolution and have GRBL output data via i2c or onewire that the second Arduino would read. Not sure is the GRBL Arduino will have enough memory left to use these libraries though.
My other option would be to port it over to a Mega2560 and use 16bit timers there.

chamnit, 2016-12-19:
@mayhem2408 : The Mega2560 would be a lot more ideal than trying to shoehorn more into the 328p version, which is completely maxed out. Over the break, I plan on updating Grbl-Mega to v1.1, so I would wait until I did that.

mayhem2408, 2016-12-19:
Another option would be to take the PWM from GRBL, feed it into another Arduino, then scale it accordingly. So I would have a smart laser controller. GRBL outputs 0-100% at 256 steps, but the second Arduino only sends out 5-15% duty using a 16bit timer or even scaling that back to 10-12 bits giving a resolution of 100-400 steps. There would be plenty of resources in the second Arduino to have LCD screens and buttons to set it up.

However, running on a Mega2560 could solve the problems. Is moving the PWM to a 16 bit timer something you would be looking at doing?

chamnit, 2016-12-19:
@mayhem2408 : It's already on a 16-bit timer, but limited to 1024 levels. I can't remember the exact reason why. I think it had to do with an OEM request.

mayhem2408, 2016-12-19:
Just thinking about it, at 16MHz clock, 1024 levels means about 15.6KHz PWM which is also smack bang of the 16KHz optimal PWM of the 50W Laser Drivers power supply.

usbcnc, 2016-12-19:
@mayhem2408 Would you mind try my port of GRBL 1.1 on STM32 target? It can have 16 bit of PWM timer. Should be enough for your requirement.

mayhem2408, 2016-12-20:
@usbcnc I do not have a STM32 at the moment. Can you recommend one?

langwadt, 2016-12-20:
@mayhem2408 Nucleo has arduino pinout and cost ~$10 
http://www.st.com/en/evaluation-tools/stm32-mcu-nucleo.html?querycriteria=productId=LN1847

usbcnc, 2016-12-20:
@mayhem2408  This is the one I use (<$2). I also bought a STlink (also < $2) to quickly program it. It can also be programmed via serial port.
https://www.aliexpress.com/item/1pcs-STM32F103C8T6-ARM-STM32-Minimum-System-Development-Board-Module-For-arduino/32721357650.html?spm=2114.01010208.3.1.RyYyuq&ws_ab_test=searchweb0_0,searchweb201602_3_10065_10068_10084_10083_10080_10082_10081_10060_10061_10062_10056_10055_10037_10054_10033_10059_10032_10099_10078_10079_10077_427_10103_10073_10102_10096_10052_10050_10051,searchweb201603_1&btsid=39642672-1ce6-46a6-9176-6bcecc240c9e

chamnit, 2016-12-20:
@mayhem2408 : Hang tight. I just did a preliminary port to the Mega branch. It compiles but I haven't tested yet.

mayhem2408, 2016-12-20:
@chamnit : I'm hanging. I've got my Mega2560 on standby ready for flashing

chamnit, 2016-12-20:
@mayhem2408 : Sorry, didn't get to it tonight. I was wrapped up trying to track down a critical bug related to alarms, homing, and an infinite loop. Theoretically Grbl-Mega should be good to go. If you are feeling adventurous, there's a good chance that it'll work fine. Otherwise, I'll have to get to it tomorrow. It's getting late.

mayhem2408, 2016-12-20:
I've uploaded it to my barebone Mega2560 and so far so good. CNC.js, LaserWeb and LaserGRBL all working and reporting OK. Next is to wire up my stepper drivers to make sure they are responding OK.

dgoadby, 2016-12-26:
My laser cutter (HPC) has a voltage control on it. I am taking the pwm output to a simple integrator to generate a voltage for the laser based on the PWM. In order to create finer control at the lower end I am going to try a simple op-amp configured to operate with a logarithmic input to output. The circuit is simple and it should give me finer control at lower power levels.

koepae, 2017-03-10:
I just came across this and its very close to my problem.

I'm looking to control welding current pulsing.
The simplest way would be to change speed in gcode but this would result in upto 20 lines of
gcode per 1 second of movement so I'm afraid it would flood the serial connection.

I'd like to add parameters to the S word to accomplish the pulsing, for example:
S100 D1 R40 L40
In above (words used just for example)
S is high welding current
D is welding pulse duration in tenths of second
R is welding pulse relation, percentage of pulse duration that is high current
L is low current as percentage of high current
This should make the S to cycle so that value 40ms 100 followed by 60ms 40.

This should not be very complicated feature to add, but my big question is where in the code
should I do it?

As already 8 bit pwm gives 0.5 amp resolution on 100A welder so spindle speed control should work
as ready starting point
Some jitter is acceptable so a state machine solution would suffice.

I'm planning to use usbcnc's STM32 port

Rgds,
koep

allstar87, 2017-10-05:
I have a 60W CO2 power supply, an HY-T60 as a matter of fact.  It has several inputs; a TH and TL lines, as kind of enable pins; if the TH line is pulled up to 5V OR the TL line is pulled to ground, it fires.  It also has an IN line, which can be controlled by 0-5V analog signal or "5V PWM f >= 20kHz".
I'm getting an unstable beam out of the tube; should I put a low-pass filter on the PWM line?

biasedlogic, 2017-10-05:
You should do two things:
1. Change the PWM base frequency. The default one is good for a basic spindle control, something about 1kHz if I remember correctly. You can bump it up to something higher for laser control. IIRC You need to edit settings in the grbl source and recompile.
2. Then you can put a low pass filter with a much shorter time constant so that you get crisp and fast power control.

ThiagoCNC, 2018-04-09:
@mayhem2408 could you send me the changed code that you have used. my laser is 2.5W with arduino nano but I could not make the change of kHz
"" I run a 5W Laser diode and my TTL laser driver did not like the default 1KHz PWM. I changed mine to 8KHz and it runs perfectly. The drivers documented frequency is 10KHz, but 8 is close enough.

I think part of the problem was the speed I am running it. If I feed at F4800 and my resolution of the raster is 10 pixels per mm, that is 800 pixels per second. This probably means that some pixels only get 1 pulse, and some get 2. At 8KHz, each pixel get 10 shorter pulses and the result is much smoother.

So to answer your question, in my situation the PWM frequency does make a difference.""

cprezzi, 2018-10-29:
@MeJasonT Grbl doesn't use the arduino analogWrite, it's highly optimized and already generates the PWM itself with a timer interrupt! There is nearly no space left in the AVR328 to add a I2C or RS485 interface. I suggest to use a Mega256 or even better a 32 bit MCU.

MeJasonT, 2018-10-29:
Hi cprezzi, I looked at cpumap.h and discovered that the output was already a bit banged signal and not analogue write.  Do you have any idea how i can step up up the frequency to provide an 50% on/off pulse providing say a 2khz variable frequency.   My plan is to read the frequency into another Atmega 328 which already has an rs485 interface attached and transmit a 2 byte value to a third Atmega328 which is receiving the RS485 and transmitting value to my variable frequency drive via i2c DAC IC.
The current PWM output only give me a value from 20 to 255.  This translates to 74 RPM per bit.  which is a huge difference in spindle speed.  I guess what im trying to ask is, I want a value with much higher resolution - the rest is a piece of cake to someone with my electronics knowledge.  Unfortunately i'm no programmer and don't mess with GRBL that way i can trust its reliability.  Better to ask the experts. Cheers J.

cprezzi, 2018-10-29:
Why do you need variable frequency? That's what the VFD is for. Ususally you feed the VFD with a analog 0..10V signal (which can also be used with a potentiometer). 
Grbl can only create a 5V PWM signal with a fixed frequency for the spindle. What you need is a PWM to 0..10V converter like this https://de.aliexpress.com/item/PWM-zu-Spannung-Konverter-Modul-0-100-zu-0-10-v-f-r-PLC-MCU-Digital/32834921874.html

cprezzi, 2018-10-29:
@MeJasonT To get a higher resolution you could probably make a PWM to voltage converter that only creates 0..5V and add 0V or 5V in series with a mosfet from one of the digital outputs like mist or flood.

MeJasonT, 2018-10-29:
ok chaps, you have missed the great big elephant.  Im using an RS485 link between my control box and the Inverter panel for the milling machine, I did try to make the point that the VFD generates a hell of a lot of hf noise.  Hence to separation via a 485 coms link.  I have looked into plastic fibre like toslink but cant transmit the pwm/RPM signal.  The receiving board attached to the vfd has an DAC ic and an opamp generating the 10v signal for the drive.  
I have managed to use the Arduino PulseIn command to raed the pwm and get a 0~1000 ish count on the board interfaced to the GRBL control board but its random at best, not really outputting consistent values.  averaging wont be clever as i get a run of values within about +/-5 the suddenly a 14000 value.
Ill see if i can create a light meter and go back to trying a fibre link, as you guys have mentioned boosting the voltage at the control board and then using that signal.  I suppose i could sniff the serial comms from the pc and transmit the required signals from there.  It would be really sweet to have GRBL control the miller RPM.  
Cheers J.

cprezzi, 2018-10-30:
I think sniffing the serial connection is a good idea. But be aware that Grbl has an internal serial buffer and planer queue. Command execution will be delayed by some lines, so you should count the returned ok's to know when the spindle command is executed.

MeJasonT, 2018-10-30:
Thanks cprezzi, good point i hadn't thought about the planer queue.

MeJasonT, 2018-11-04:
cprezzi, ive sussed out that i can sniff the verbose response using another arduino.  I set up software serial and got the messages being output.  i also saw the advantage of it showing the spindle speed as set in the gui (ie 7500 RPM appears as 7500) it switches as soon as the grbl gui shows the on/off function of the spindle.  Now for the issue - GRBL outputs 115200 baud and the second arduino is set to communicate via software serial at the same speed - its bloody asking a lot.  obviously some of the data is fragmented.  Im wondering if altering the software baud rate would bring it into sync - any thoughts.  The next experiment is to remove the uP IC from the second arduino board (using the board as a TTL to USB converter) and see what the data looks like then.

cprezzi, 2018-11-05:
Sure you could reduce the baudrate that Grbl uses, but this would also slow down the speed you can send commands to Grbl, which could lead to stutter. 
I would use hardware serial for this connection and software serial for the RS485 connection (less traffic).

MeJasonT, 2018-11-05:
Ok Buddy, huge success sort of.  I can sniff the messages being sent from GRBL back to GUI listening in on the TTL TX pin.  works a treat and does not affect GRBL.  Now all I have to do is parse the string and split out the bits i want (that,s the sort of bit).  whats more the GUI is sending the desired RPM and not the PWM - that must be handled in the spindle code, perfect for me.  Once i work out hot to parse the string i can then use the sniffer Arduino to create and send the RS485 messages to the remote device.

(Just used a serial echo to bang out what was received from the GRBL TX pin)

<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,1000|Ov:100,100,100|A:SF> 
FS = feed, speed
A:SF S or C =spindle fwd/rev  F= suds

so im looking for delimiters FS: and taking the second set of values for spindle speed
(FS:0,1000) where the first is the feedrate
Then A: as the second delimiter and doing a case/switch on the characters to get the direction and suds/coolant commands.

If you felt like writing some example parse code for me it wouldn't go unappreciated and would certainly speed things up for me. 

I have discovered the the same could be used to interface a Torch Height controller to my CNC Plasma table (also on GRBL).  Sonny is a star, if I ever get to meet him in person he will be so drunk on the beer I owe him.

I was originally thinking that i needed to control the RPM for cornering and HSM machining techniques but working on this Inverter interface i have discovered that the RPM stays constant and the feedrate is adjusted for cornering etc.  I guess that's how its meant to be, so the way i'm currently running the machine by manually starting and stopping the spindle and setting the speed isn't so bad.  Just a pain when i need to stop for a crash or whatever.  will keep you updated on progress.  J.

cprezzi, 2018-11-06:
Be aware that you could also override the spindle speed by 10-200% which is reflected by the Ov: response. You probably need to watch that too.

Do you know "laser mode" ($32=1)? This could eventually be helpfull for your plasma cutter. See https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode for details.

MeJasonT, 2018-11-06:
Th plasma is working 100%, I have not bothered with the laser mode. its basically 0 or 255 PWM o/p and the suds command for pilot and arc ignition.  The Torch goes down to 0.5mm of the sheet metal ish. the suds is enabled - enabling the pilot relay (a 300A Motorbike starter relay) this energises for 1.5 to 2 seconds.  During this time the ark is struck by switching on the spindle (Ark Trigger).  The pilot is then disengaged as hopefully by this time the plasma has been created and maintains the flame - not failed yet, apart from on really shitty steel.  Ive had the odd flame out when working close to the edge of the material as you would expect.  The High freq noise at this point tends to kill GRBL.  Best to keep inside the material edge, i now keep a 5mm border around work.  all of this is operated withing the Cambam Post Processor.  You know when something is working it gets left alone, i'm kind of there tight now. 
As for Spindle overspeed i will make sure to limit the max speed to 7500 RPM so if it does get higher commands it will only send the max speed command.  In the unlikely event that the DAC on my remote micro receives a higher max value it can only produce 0-10v via its opamp - the inverter has this mapped as 0 ~7500 RPM.  what actually happens is the DAC rolls over its only 12 bit I2C dac ? (need to check facts) it ends up back a zero and ramps up from there.  Inverter wont go over 7500 RPM it physically cant unless i get back into its setup and increase its frequency output range over 100hz.  The 3 phase motor driven at 50hz output frequency gave 3600 RPM max for straight forward 3 phase motor (via belt drive).  The manufacturers of the VFD have advised not to exceed 110Hz as the torque of the motor starts to diminish.  So 100hz it is.  Crap for Aluminium cutting with small tools <3mm.
I have tried for 24 hours now to parse the serial input spring -no luck yet I'm one of those copy and paste programmers i'm afraid otherwise known as a mechanical engineer.   Cheers  J.

MeJasonT, 2018-11-06:
The plan for THC is to use a capacitive or laser sensor to calculate height.  get the activation commands the same as  i'm trying to achieve now with the serial snoop. and using the command to bypass or control the Z stepper motor.  I'm not convinced i even need one, unless i'm going to cut up inclines or corrugated surfaces, like never.  i need to get on and make chips really instead of all these gota have niceties.

GOodGOD1, 2018-11-19:
@chamnit hi and thank you for your effort. Have you released the 1.1f for mega already? Or is it the docwelsh the most recent? Thanks

mlinka, 2019-02-17:
@mayhem2408 Can you share your hex file with 8khz pulse frequency. I have 15W laser with Elekslaser GRBL firmware and I think 8Khz would work better for engraving stainless steel. Thank you. Or is there a different way to recompile GRBL because I cant use Arduino IDE because Nano has limited memory.

amoineau, 2020-11-13:
Hi,

I'm jumping in to check if the frequency switch feature has been implemented. I tried to follow the different threads but couldn't find the answer. I'm sorry if I missed it in this (very long) issue !

I'm using a relay to divert the PWM from one tool to a second but they are not using the same frequency. I use M7 and M9 to set/reset the relay. So I'd like to have 2 frequency presets and be able to switch between them when using M7/M9.

If the solution is not already available I don't mind modifying the source code myself but I could use some pointers on what to do / not do.

Thank you in advance !