gnea/grbl #59: Jog Motion #2

quangvi, 2016-12-09:
no , Hi @chamnit Feedrate override do not work with jog motions, but problem is happen, if the axis had move slow speed , i want to change fast speed , i only press stop or reset,

chamnit, 2016-12-09:
Please read and try to understand the jogging documentation in the doc folder. It explains how it works.

closed 2016-12-09 by chamnit

quangvi, 2016-12-09:
Hi @chamnit  , I understand is work with "manual command" but the button "X+X-,,Y+ Y-,Z+,Z-. not accept feedrate overide, This is intentional?

chamnit, 2016-12-09:
No, I don't believe you do. Grbl has no "buttons". You are confusing a GUI function for Grbl's. I'll say this again. Feedrate overrides are ignored for jog motions and please try to read and understand the jogging document. It's all explained there.

quangvi, 2016-12-27:
thank @chamnit,  sorry my english a little stupid , I'll edit code GRBL part Jog, it will accept buttons x + ,y +, z +, x-, y- ,z- can press feedrate overide, in my machine , i had designed buttons chance in your  GUI function for Grbl's,

quangvi, 2016-12-27:
my machine have 1000mm for Z axis or more, 30mm for X axis or more, it,s lathe machine cnc 2 axis, axis so long travel, if i press X+ or Z+ in Gui or button (my additional ) with feedrate low, i .will watting along time,it's fidget

quangvi, 2016-12-27:
thank @chamnit,  sorry my english a little stupid , I'll edit code GRBL ,Feedrate override will  work with jog motions, it will accept buttons x + ,y +, z +, x-, y- ,z- can press feedrate overide, in my machine , i had designed buttons chance in your  GUI function for Grbl's,

chamnit, 2016-12-27:
@quangvi : Please understand that feedrate overrides do NOT work with jogging. This is completely intentional. It prevents misuse and accidental damage when exiting the jog state and starting a program. The override value may have been set very high, forgotten, and result a hard crash into the material. This usually results in a broken bit, damaged spindle, ruined workpiece, and even machine damage. If you want it to work with jogging, please place it in your own repo. This will not be added to Grbl.

quangvi, 2016-12-27:
thank you @chamnit  , I understand ,   in many cnc machine industrial (I have machine cnc 4 Axis)  , feedrate overrides is work with jogging, even  feedrate overrides work  homing cycle . the  results in a broken bit, damaged spindle, ruined workpiece, and even machine damage had warning in user manual ,users must be responsible for the misuse or forget ,because Z axis not returned home(or raised off the planer) , X axis or Y axis moving will broken drill , it is the user's fault , i will place it in my own repo thank .

chamnit, 2016-12-27:
@quangvi :

> in many cnc machine industrial (I have machine cnc 4 Axis) , feedrate overrides is work with jogging, even feedrate overrides work homing cycle .

Allowing feed overrides during these tasks makes absolutely no sense. You have direct control over  jogging feed rate if the host GUI implements it properly, and everyone knows that altering homing rate can change its repeatability and accuracy. Either way, it's dangerous and lazy.

quangvi, 2016-12-27:
thank you @chamnit , I understand , but workers use it and they do not want to change, their programming  Gcode directly on a 7-inch LCD interface, and feedrate overrides work with jogging, ex : they enter Gcode
G1 X20 Z20 F100 but true Gcode is G1 X20 Z20 F200 ,this call trial Gcode in jogging, they don't want waiting or press stop button and want press feedrate overrides 200% , this is reason

chamnit, 2016-12-27:
@quangvi : I still don't see what your problem is. `G1 X20 Z20 F200` is valid g-code. Grbl would update the motion with feedrate overrides. This is not a unique jogging motion, like with a rotary dial or jogging increment keypresses on an industrial machine. We are talking about two totally different things here.

If you choose to jog with G1, that's your preference. Grbl's new jogging mode is designed for usage that is completely independent of g-code parser and machine state. Again, it's up to the GUI to handle how responsive their jogging is to rate changes. I've outlined how to do this in the jogging wiki page, like I mentioned for you to read and study.

quangvi, 2016-12-27:
nono , this is call " try Gcode in JOG" , the workers( in my company ) , 1 st ,they program Gcode  G1 X20 Z20 F100 ( JOG ) ,2st they will press button Feedrate overide up or down until cushy, ex: the Gui is display 200% now , Gcode relapce in File cnc is G1 X20 Z20 F200

chamnit, 2016-12-27:
@quangvi :  I don't give up often but I'm giving up trying to explain this. Grbl's feed rate overrides work with any normal gcode command, like a G1 as you have described. I don't know how many other ways I can say this. Whatever you are doing that is causing to not work is not a Grbl problem. It's your host GUI. Grbl is only a service. YourGUI is responsible for sending gcode and override commands to Grbl. Please just stop.

quangvi, 2016-12-27:
example 2 : they press button X+ , X- or more(Y or Z) , Feedrate overide up or down will accept ,because work "Try Gcode" ex: they want  X axis move positive 100mm but they waiting  30 second(in JOG) they not change feedrate , they only press stop button, they want chance feedrate and waiting 15 second (in JOG)

quangvi, 2016-12-27:
sorry @chamnit , that is  exchanged for better GRBL, i will edit source GRBL in my own repo thank