gnea/grbl #60: Probe cycle problems

109JB, 2016-12-10:
I am writing some automated probing cycles for my GUI and have come up against a problem.  I am not sure if this is a GUI or a Grbl problem, but I believe it is Grbl.  The probe cycle I am implementing is a corner probe cycle where it goes through the many steps. Below is an example of a few lines where the problem occurs, with the grbl echoes shown in bold. I have added a description of what the code is doing with each issued command.

==== START Z ====
**[echo: G91G38.2Z-0.3F1]** (Probe Z- (faster feed rate))
**[PRB:-3.0000,-3.0000,-3.0375:1]**
**ok
[echo: G91G0Z0.100]** (rapid Z+ for pull off)
**ok
[echo: G91G38.2Z-0.2F1]** (probe Z- (slower feed rate)
**[PRB:-3.0000,-3.0000,-2.9454:1]
ok
[echo: G91G0Z0.100]** (rapid Z+ for pull off)
**ok**
==== Z Done ====
==== START X ====
**[echo: G91G0X-0.3]** (rapid -X to get off corner)
**ok
[echo: G91G0Z-0.2]** (rapid Z- to position probe below top of part)
**ok
[echo: G91G38.2X0.3F1]** (probe X+ (faster feed rate))
**[PRB:-3.2879,-3.0000,-3.0468:1]
ok
[echo: G91G0X-0.100]** (rapid X- for pull off)
**ok
[echo: G91G38.2X0.2F1]** (probe X+ (Slower feed rate))
**[PRB:-3.3772,-3.0000,-3.0468:1]
ok
[echo: G91G0X-0.100]** (rapid X- for pull off)
**ok**
==== X Done ====
==== START Y ====
**[echo: G91G0Z0.2]** (rapid Z+ above part)
**ok
[echo: G90G53G0X-3]** (rapid X to where X started from)
**ok
[echo: G91G0Y-0.3]** (rapid -Y to get off corner)
**ok
[echo: G91G0Z-0.2]** (rapid Z- to position probe below top of part)
**ok
[echo: G91G38.2Y0.3F1]** (probe Y+ (faster feed rate))

There are more steps after this, but this is an example of where a problem is occurring.  As you can see, Grbl echoed the G91G38.2Y0.3F1 line properly, but never starts movement on this command. Also, there is never a probe response sent or an "ok" message, or Alarm, or error. When this happens, my GUI is waiting for a probe result. The problem doesn't always occur at this specific spot, but it always happens at the issuance of a G38.2 command. Sometimes it gets all the way through the whole subroutine successfully. At the point or failure, the GUI is still receiving status responses, and they show that Grbl is Idle. Here is a cut and paste of one. 

<Idle|MPos:-3.4142,-3.0000,-3.0468|FS:0.0,0|Ov:100,100,100>

While in this state, non-realtime commands don't work, but I can send a reset an it resets, but something is going on.

Let me know what you think.

chamnit, 2016-12-10:
@109JB: First can you provide your $$ settings, build info string, and Arduino IDE version you compile with? 

Second, can you post this problem program so I can try to reproduce the issue? 

Third, have you ruled out external influences like your GUI or machine be setup by running the another GUI or machine?

109JB, 2016-12-10:
1. Of course.  The $I and $$ info is below. I am using Arduino IDE 1.6.12.

2. Not sure what you mean by "the problem program" The GUI uses user inputs, assembles lines of G-code to send to GRBL, and sends them. The actual lines of code that were sent to grbl are in the first post from the GRBL echoes.  Below the $$ settings, I have pasted another set of grbl echoes further below. The last line is the one it hung on this time, which was a different G38.2 command from the first post.

3. As far as external influences, there shouldn't be any.  I am not testing on the machine yet, My current on an UNO clone with the 16U2 USB chip and a pushbutton wired to the probe pin on the uno.

A little explanation of some of my GUI stuff.  When a [PRB: response is received from grbl, my serial port parser goes to work, the axis coordinates are stored in a double precision array, and another integer variable is used to store the pass/fail bit.  I am using an integer for the pass fail rather than a boolean because I set this to a value of 3 to denote that the probe cycle was started. ie: 0=fail, 1=pass, and 3=probe started. Additionally there are user defined values for the max probe distance, and probing feed rate. So, for the portion of code involving a probe I do this:

-----Get the user defined ProbeDirection, ProbeDistance, and ProbeRate-----
ProbeResult = 3
SerialPort1.write("G91G38.2" & ProbeDirection & ProbeDistance & "F" & ProbeRate & vblf)
Do While ProbeResult =3
    Application.DoEvents()
End Do
-----Get the probe results -----
-----Do other stuff -----

I have confirmed that the program is inside the Do loop waiting for a probe result to be received from grbl when this happens.

=======================================

[echo: $I]
[VER:1.1e.20161204:]
[OPT:VM]
ok
[echo: $$]
$0=10
$1=255
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=1
$20=1
$21=1
$22=1
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=250.000
$101=250.000
$102=250.000
$110=1000.000
$111=3000.000
$112=4000.000
$120=200.000
$121=200.000
$122=200.000
$130=200.000
$131=200.000
$132=200.000
ok

===========

[echo: G91G38.2Z-0.3F1]
[PRB:-3.2943,-3.6809,-2.6213:1]
ok
[echo: G91G0Z0.100]
ok
[echo: G91G38.2Z-0.2F1]
[PRB:-3.2943,-3.6809,-2.5233:1]
ok
[echo: G91G0Z0.100]
ok
[echo: G91G0X-0.3]
ok
[echo: G91G0Z-0.2]
ok
[echo: G91G38.2X0.3F1]

chamnit, 2016-12-10:
@109JB : I meant post the gcode program with this problem, so I can run it on my test rig. 

As far as ruling out the GUI, this is the first thing I try when debugging streaming issues. I  usually try to test against a streamer that uses a different serial comm library and different OS if possible. It ensures which side of the stream the problem is on. From what it sounds like, it's Grbl but I would test this anyway.

109JB, 2016-12-10:
@chamnit - OK.  The g-code that would be sent would be as shown below.  The code sometimes gets through, and sometimes hangs waiting for the probe response from GRBL right after sending a G38.2 line.  At this point the buffer should be empty because the GUI is waiting before sending the next line. I'm going to do some more testing on my end too.  

G91G38.2Z-0.3F1
G91G0Z0.100
G91G38.2Z-0.2F1
G91G0Z0.100
G91G0X-0.3
G91G0Z-0.2
G91G38.2X0.3F1
G91G0X-0.100
G91G38.2X0.2F1
G91G0X-0.100
G91G0Z0.2
G90G53G0X-3
G91G0Y-0.3
G91G0Z-0.2
G91G38.2Y0.3F1
G91G0Y-0.100
G91G38.2Y0.2F1
G91G0X-0.100

chamnit, 2016-12-10:
@109JB : Been testing this with the G38.3 command, rather than G38.2, because it won't alarm if the probe fails. I don't have a switch to simulate a probe trigger. So far, I haven't had a problem. Does your job behave differently if you switch over to G38.3?

109JB, 2016-12-10:
I'll check.  Getting ready for the office Christmas party though, so may not be til tomorrow.

109JB, 2016-12-11:
Had some time to check some more.

G38.3 does not seem to make any difference

As I said this is for a automated probing routine. As such, there are many points where linear G0 commands come immediately before the G38.X command.  I discovered a few things during my most recent testing.

1. When the problem occurs it acts like Grbl is in a Hold state, but Grbl is not reporting Hold in the status reports. However, if I send a "~" command Grbl resumes motion and acts fine. 

2. Since this is an automated routine with linear G0 moves immediately before the G38.X command, I found that if I have the GUI wait for grbl to be Idle and then an additional 100ms delay before sending the G38.x line that I can't get the problem to occur.  So it does this:

Send linear G0 command
Wait for "Idle"
Wait an additional 100 milliseconds
Send G38.x command

Not sure if this points in a certain direction or not but will keep testing.

chamnit, 2016-12-11:
@109JB : Thanks. This is a pretty big clue. This may mean that the state machine handling of the probing cycle exit may be a little off. BTW, I'm ran my tests without a probe, so they would always reach the end of the probe and fail. Meaning I didn't switch anything to cause Grbl to stop the cycle. Are you engaging the probe switch for the G38.3 test?

109JB, 2016-12-11:
Yep.  I'm engaging the switch by watching the DRO and the grbl echo responses to see when it is probing. 

After doing more testing, It looks like it is not stopping in the probe cycle, but either right after a probe cycle, or as soon as another probe cycle is read, Grbl is going into a Hold type state without any notification to the user that it is in a hold.  Here is what I have found:

G91G38.2Z-0.3F1  --------------- Completed fine
G91G0Z0.100 --------------- Completed fine
G91G38.2Z-0.2F1 --------------- Completed fine
G91G0Z0.100 ----------------This move starts but stops part way through as if put in feed hold
G91G0X-0.3
G91G0Z-0.2
G91G38.2X0.3F1 ----------------- This is the last command that was sent to GRBL and was echoed
G91G0X-0.100
G91G38.2X0.2F1
G91G0X-0.100
G91G0Z0.2
G90G53G0X-3
G91G0Y-0.3
G91G0Z-0.2
G91G38.2Y0.3F1
G91G0Y-0.100
G91G38.2Y0.2F1
G91G0X-0.100

It doesn't always stop there, but always stops in the move right after a G38.x command, and the next G38.x command is the last command that GRBL echoes back.  As I said, the machine is essentially in a hold state, and sending a "~" command restarts it. When it is this suspected feed hold state, the status reports show "Idle" and not "Hold. Very puzzling.

109JB, 2016-12-12:
I did some testing on the machine and with a programmed pause just before the G38.x command all works fine.  The pause first waits for GRBL to become "Idle", and then waits an additional 200 ms before issuing the next G38.x command. I ran about 100 probing cycles this morning and all worked fine with the pause in there.  

I didn't think to do it while I was out in the shop, but later, I will comment out the pause and see if it still has the problem.  I was thinking that perhaps just using a pushbutton for testing could have something to do with it. I will report back what I find without the pause.

chamnit, 2016-12-12:
@109JB : Awesome. Thanks. Your enforced pause results suggests that this error would only occur with multiple, sequential probing cycles. Another good clue. I'll delve into the code later today to see if I can find anything. BTW, have you checked v0.9j to see if it has the same issue?

chamnit, 2016-12-12:
@109JB : I have my portable test rig with me today. During lunch, I wired a switch to the probe pin, flashed v1.1e to it, and wrote your settings (turned off soft limits though). I've been streaming your example g-code below, each test loops 3 times, with the send-response protocol. (Tried character counting too). I can't get Grbl to lock up like you have described after almost 20 tries. Anything I'm missing?

```
G91G38.2Z-0.3F1
G91G0Z0.100
G91G38.2Z-0.2F1
G91G0Z0.100
G91G0X-0.3
G91G0Z-0.2
G91G38.2X0.3F1
G91G0X-0.100
G91G38.2X0.2F1
G91G0X-0.100
G91G0Z0.2
G90G53G0X-3
G91G0Y-0.3
G91G0Z-0.2
G91G38.2Y0.3F1
G91G0Y-0.100
G91G38.2Y0.2F1
G91G0X-0.100
```

109JB, 2016-12-12:
Haven't checked 0.9j.  The GUI I am working on is a new one that was written from the ground up for V1.1. It doesn't support the old style status reporting.

chamnit, 2016-12-12:
@109JB : Can you try the Grbl stream.py script? I've been re-running the g-code program off and on today, but still doesn't hang.

109JB, 2016-12-13:
@chamnit : I wasn't able to get back into the shop tonight, but I did some more testing with my GUI and test rig.  I forgot that I had set the baud rate to 9600 because I had been testing some things with serial communication.  To test if the baud rate setting could have anything to do with it,, I reset to 115200, and eliminated the artificial pause and ran about 50 test cycles on my test rig without problem.  I then reset it back to 9600 baud and had the problem occur on the second run through the cycle.  So it appears the baud rate can somehow trigger this. At 115200 I have not been able to replicate the problem.  My test rig is a clone UNO with the 16U2 chip, but I don't think this one has the upgraded firmware that Alex Holden posted. 

It is odd though that the it acts like a feed hold without "Hold" in the status response.  I know that I am still getting status responses streamed as I have a text label that updates with every status response and can see the additional information flashing through.  Is there something that could cause GRBL to go into feed hold without updating the state in the status responses?

109JB, 2016-12-13:
@chamnit : Forgot to mention that I don't have Python installed on my computer. I will see about doing that tomorrow.

chamnit, 2016-12-13:
@109JB : Testing with Python probably won't be necessary. I think I know what the issue is. The serial TX blocks other processes if it has to wait to place characters in the TX buffer. I think it's the combo of running at 9600 baud, enabling echos, and you likely having reports. This probably caused the probe loop to expire when the serial TX sat and waited for too long. I assumed the baudrate would always be 115200 or greater and designed the reports to never fill the TX buffer at that rate. Fixing this potential TX buffer has been on the todo list for a while but hasn't brought up an issue until now. If this is true, I still think it's safe to say this isn't an issue I need to fix immediately.

109JB, 2016-12-13:
@chamnit : Sorry I forgot about the 9600 baud setting.  I think I will make a note in the WIKI known bugs to let people know that running at too low a baud rate could result in unpredictable behavior.  I guess I wouldn't really consider this a "bug", more like a limitation that can be lived with since normally 115200 will be used anyway.

109JB, 2016-12-13:
@chamnit : On the WIKI note, I went to add a short note, and the WIKI page must be locked from user editing. Not sure if this is intentional or not.

chamnit, 2016-12-13:
@109JB : Thanks. The Wiki write access is fixed. I'll also put a note in the config.h file as well.

closed 2016-12-13 by chamnit