gnea/grbl #61: support servo as spindle (marge request in fact)

Cdriko, 2016-12-12:
Hello
I want to use a servo in place of a spindle (for a vinyl cutter/plotter).
I find this good work made by 
[robottini](https://github.com/robottini/grbl-servo)
is it possible to merge it to the master branch?

thanks

chamnit, 2016-12-12:
@Cdriko : Perhaps in v1.2. Not this recent release. I'll put this link in my todo notes and close this issue.

closed 2016-12-12 by chamnit

langwadt, 2016-12-13:
It would also be useful for using an RC brushless motor and speedcontroller for a spindle

Cdriko, 2016-12-13:
good idea .
but for the moment I find a solution with a L298 and a solenoid
thanks

sbma44, 2017-02-08:
I've just built a 4xidraw and would love to get onto a modern version of grbl for the fixed corexy homing behavior, but servo support is a blocker. Would you accept a PR for this functionality?

Danccls, 2020-02-01:
I'm testing a servo on the spindle. Later I want to use a brushless RC motor ans a ESC as spindle.

The problem is that maximun speed must be configurated to 255, and the spindle selector moves betwen 0,100 or 200. Actually I only have two speeds.

I can use the command M3 Sxxx (where xxx is the speed of the spindle from 0 to 255), but it will be usefull the ability to select the steeps of spindle selector, as 5 or 10, not the present 100.

Regards.