gnea/grbl #62: Limit problem

Marco70cn, 2016-12-13:
Hello, i tried everything but can't use limit XYZ. The pin 9,10,11 connected to ground directly or trought a resistor from 5V have no results. The engine continue to run as normally. Actually i using Grbl0,9J because the new release 1.1 seem like not working on Universal Gcode Sender V1.09
The CNC run but no feedback about axis position.
However, the problem most important is how to use the limit switch using arduino uno.
Thanks in advance, Marco

109JB, 2016-12-13:
What are your $$ settings?

chamnit, 2016-12-13:
@Marco70cn : The z limit is on pin 12 as shown on the diagram of the connecting Grbl wiki page.

Marco70cn, 2016-12-14:
@chamnit Not Z limit problem, now i only working on X axis, but impossible let recognize the limit switch to arduino.

@109JB My $$ settings:
**** Connected to COM6 @ 115200 baud ****

Grbl 0.9j ['$' for help]
>>> $$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=400.000 (x, step/mm)
$101=400.000 (y, step/mm)
$102=200.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)
ok

109JB, 2016-12-14:
You don't have hard limits turned on. You need $21=1 before the limits will work.

Marco70cn, 2016-12-14:
Thank you very much. Now it's working.

closed 2016-12-14 by Marco70cn