gnea/grbl #63: What happened to the ENABLE_SOFTWARE_DEBOUNCE?

trustfm, 2016-12-13:
What happened to this ? It is simply removed ?

chamnit, 2016-12-13:
@trustfm : Yes. It was removed, but easy to put back. Over the years, there has been no indication that anyone used this. My argument against having it is that if you have noise in your limit pins, you will eventually have a false reading. The software denounce only masks the issue. It doesn't solve it. 

That said, I originally removed it so I can free up the watchdog timer to detect timeouts. That feature didn't make it into v1.1. If there are enough users asking to put back software denounce, I can accommodate that.

trustfm, 2016-12-13:
This feature is fundamental if you have moving endstops in critical places. 
I have made machines based in this feature. 
Take a look at this video : http://www.conversioncenter.net/pickandplacer.mp4

closed 2016-12-20 by chamnit