gnea/grbl #65: S0 doesn't turn off laser

T2Laser, 2016-12-14:
When VARIABLE_SPINDLE is disabled, S0 has no effect and the laser stays on.

In 0.9 it didn't matter if PWM was used or not S0 still had the same effect as M05.

chamnit, 2016-12-14:
@T2Laser : Grbl v0.9 also did not disable the spindle when VARIABLE_SPINDLE was disabled. See https://github.com/grbl/grbl/issues/933. 

Laser mode is not compatible without VARIABLE_SPINDLE anyway. You get no benefit running Grbl this way. May I ask why you need VARIABLE_SPINDLE disabled with a laser?

closed 2016-12-14 by chamnit

T2Laser, 2016-12-14:
Some boards are hardwired for D12 and/or don't support PWM.

It would be nice if the same code worked on both configurations. 

Will using M05 in it's place have any negative effect on laser mode performance, particularly when scanning (pausing etc)?

chamnit, 2016-12-14:
@T2Laser : The support PWM part is easy to fix. Just program spindle speed `S` to your max rpm. That's it. `S0` should also work fine in this mode. 

As for older boards hardwired for spindle enable on D12, I'm going to have to pass on this. VARIABLE_SPINDLE was introduced a couple of years ago and is the standard build. It's going to take a lot more work than I'd want for laser mode to work with VARIABLE_SPINDLE disabled. It was designed specifically to work with it. At some point, I just have to say no. It's not worth.

I just checked the settings source. You shouldn't be able to enable laser mode at all without VARIABLE_SPINDLE enabled. It should return a disabled message if you try to program $32=1. I will also add a compiler check if the laser default is enabled when VARIABLE_SPINDLE is disabled in the next push to prevent users from using laser mode without it.