gnea/grbl #66: Grbl state during G4

109JB, 2016-12-14:
Grbl reports "Idle" during a G4 pause command. There are several reasons why I personally believe this is wrong, but the primary one is safety. 

A user that sees an idle state might think that the controller is really idle, which it is not in reality, and if commands from whatever GUI are slated to be sent, the machine could suddenly start moving unexpectedly.

Another reason I feel this is wrong is because even though a G4 is not a motion command, it is still a valid part of a G-code program. Other controllers enter a Run state and don't exit that state until an program end command such as M02 is encountered.  I understand that GRBL operates differently since it operates on a single line at a time, but if a line is being operated on, even if it is a non-motion line, shouldn't Grbl show "Run"?

arkypita, 2016-12-14:
Already asked: https://github.com/gnea/grbl/issues/54

closed 2016-12-14 by chamnit

109JB, 2016-12-14:
Sorry. I missed the previous post on this subject.