gnea/grbl #69: Homing Problem with V1.1e

bvernham, 2016-12-20:
I have been running V0.9J for quite a while with no issues.

Today I flashed V1.1e with the same options as I had set in V0.9j.

I reset and unlocked and moved X/Y/Z and confirmed directions are correct.

I then homed the machine.  Z homed as expected but as soon as the x/Y started to home the machine locked.

bCNC sometimes indicated and alarm (but not a specific #) and other times indicated idle.

I thought at bCNC may be causing some issue so I just tried a serial terminal and homed.  After the Z finished homing, the Uno locked up and would not respond to anything.  I tried to unlock "$X" and nothing.

Any ideas how to diagnose the problem?

Thanks

Bruce

chamnit, 2016-12-20:
What are your $$ Grbl settings and $I build info string?

bvernham, 2016-12-20:
$IGrbl 1.1e ['$' for help]
[MSG:'$H'|'$X' to unlock]
[VER:1.1e.20161218:]
[OPT:VL] $$$0=10
$1=25
$2=0
$3=3
$4=0
$5=0
$6=0
$10=19
$11=0.020
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=3
$24=25.000
$25=250.000
$26=250
$27=1.000
$30=12000
$31=4000
$32=0
$100=40.040
$101=40.014
$102=188.976
$110=5000.000
$111=5000.000
$112=500.000
$120=250.000
$121=250.000
$122=50.000
$130=-290.000
$131=-290.000
$132=110.000
Thanks Bruce

&mdash;
 
How To Fix Your Fatigue (Do This Everyday)
gundrymd.com
http://thirdpartyoffers.juno.com/TGL3131/5858a1edcee6f21ed3d30st04vuc

chamnit, 2016-12-20:
@bvernham : I found an issue with alarm handling, particularly when dealign with homing cycle failures. It wasn't show the correct code and entering an infinite loop. This might be related to your issue. Can you please download and try the latest version? It should fix your problem, or at least tell you why it failed.

bvernham, 2016-12-21:
I downloaded, duplicated the configuration with the previous commit and re flashed.

I got the same results.

The z homed with no problem but again it locked trying to home the X/Y.

Just to make sure I was not losing my mind I flashed back to the V0.9J.

Works like a charm.

Homes correctly with terminal program and it is happy with bCNC so I am not sure why V1.1e is not working with my set up.

Ideas to try?

Thanks

Bruce

chamnit, 2016-12-21:
Why are your max travel settings negative?

bvernham, 2016-12-21:
According to the bCNC wiki, this is the only way to "fix" the workarea orientation:"Hi @yOPERO it is a common convention that the workspace is at negatives coordinates look at the #42. You have to set the workarea dimensions in the configuration of bCNC as well in grbl. After homing it should display correctly"

Do you think this is causing the heartache?

Thanks

Bruce

chamnit, 2016-12-21:
It could be. I'm surprised you were able to do it somehow. The '$' settings have a check to prevent negative numbers. Perhaps you did it through defaults instead. If you need to change where the origin is set by Grbl, enable the HOMING_FORCE_ORIGIN option in config.h instead.

bvernham, 2016-12-21:
I used the HOMING_FORCE_ORIGIN in 0.9J and I will try in V1.1e tomorrow, setting the travel max's to positive and give it a try.

But there have been multiple references to setting the max travel setting to negative so if this is the causing issues then I think this needs to communicated as something not to do.

Thanks

Bruce

chamnit, 2016-12-21:
Yes. That's what the negative check does when you enter a $ setting by normal means. But I will close this workaround with some compile time checks of the default file.

bvernham, 2016-12-21:
I would also suggest to make a note in the Wiki configuration tab.

chamnit, 2016-12-21:
Feel free to edit it, as it is a Wiki.

bvernham, 2016-12-22:
OK, max travel values can not be negative.  Changed the defaults to positive and home works as expected.

Thanks

Bruce

chamnit, 2016-12-22:
@bvernham : Thanks for reporting that it worked. Compiler checks should help provide better feedback to avoid this problem in the future. It'll be in the next push.

closed 2016-12-22 by chamnit