gnea/grbl #71: Spindle DIR as Enable Acting differently in V1.1e

bvernham, 2016-12-22:
I am using the option #define USE_SPINDLE_DIR_AS_ENABLE_PIN to "wake up" my spindle controller.

When this pin goes high (M3) I use it to turn on the power supply (48V DC) and in anticipation of the speed request (S####).

With V1.1e, when issue the M3, nothing happens until I issue the S####.

Can this be changes so that the M3 will set the DIR pin high so my spindle controller can turn on the power supply as with V0.9J?

Thanks

Bruce

chamnit, 2016-12-22:
@bvernham : Thanks. I'll take a look at it when I get a chance. This feature was one of the things I overlooked to test.

fl0yd, 2017-01-03:
@chamnit is it as simple as making the rpm have a default value of 0.0?

```
// Sets spindle running state with direction, enable, and spindle PWM.
  void spindle_set_state(uint8_t state, float rpm);
```

Changes to

```
// Sets spindle running state with direction, enable, and spindle PWM.
  void spindle_set_state(uint8_t state, float rpm = 0.0);
```

chamnit, 2017-01-03:
@fl0yd : Nope. But I just pushed a fix. I'll make a release whenever someone reports back that it works. Please make sure to test for the USE_SPINDLE_DIR_AS_ENABLE_PIN option turned off too. If you have a laser, also test with this. Thanks.

fl0yd, 2017-01-03:
Thanks @chamnit we'll have to see what @bvernham's test show.

I was looking at the commit and still don't see how passing in a null value for `rpm` will set the pin high on strictly an `M4` without an `SXXXX` value.  Though my familiarity with the code started today... :)

chamnit, 2017-01-03:
@fl0yd : Look at the source code file, not the commit. It'll make more sense there.

bvernham, 2017-01-04:
Gentlemen, the difference is with reaction to the M3 command with or without and SXXXX value < the minimum configured value. Thanks Bruce

&mdash;
 

Affordable Wireless Plans
Set up is easy. Get online in minutes.
Starting at only $14.95 per month! 
www.netzero.net?refcd=nzmem0216

chamnit, 2017-01-04:
This is what I fixed in the last push. Please test. Thanks.

chamnit, 2017-01-07:
@bvernham : Were you able to test this fix?

bvernham, 2017-01-07:
Not yet. I went back to 0.9J because 1.1e does not work with inventables easel.

 
Warning: Don't Use Probiotics Before You See This
Gundry MD
http://thirdpartyoffers.juno.com/TGL3131/5870427dcf9b1427d4659st01vuc

chamnit, 2017-01-07:
@bvernham : FWIW, I met with Inventables and some of their Easel devs over the holiday break. They were quite busy with other higher priority tasks, but they are still planning to update to v1.1's new protocol AFAIK. I notified them back in October to get it into their queue.

chamnit, 2017-01-11:
@bvernham : I am assuming that this is fixed unless you report otherwise. Closing.

closed 2017-01-11 by chamnit