gnea/grbl #72: laser mode override / procedural question

PeterGrace, 2016-12-22:
I have loaded 1.1 on my MPCNC and have tested laser mode with it.  So far it does exactly what I'd hope it to -- only engage on cut moves, so I'm super excited.  Thanks Sonny!

I do have a procedural question or maybe a feature request:  When setting up my laser, I could not spindle override and bring the laser online to do focusing activities.  I had to turn off laser mode to be able to enable the laser with it stopped.  

Admittedly it's only two extra commands so it's not a crisis that I have to do this, but I was wondering if it's possible to have the spindle pin activate under certain conditions?  Perhaps allow laser to be enabled during a Dwell command?

Again, thanks Sonny for this laser mode feature, it makes my cam processing of laser etches so much easier!

chamnit, 2016-12-22:
M3 along with being in G1, G2, or G3 motion mode states will keep the laser on. It's mentioned in the laser wiki page.

closed 2016-12-22 by chamnit

PeterGrace, 2016-12-22:
Yes, but I wanted the laser to be stationary so I could focus the beam.  Can you issue a G1 without it actually having a motion specified?  I hadn't thought to try that.

arkypita, 2016-12-22:
I agree with @PeterGrace enhancement request.
Laser focusing in 1.1 is an annoying task when "laser mode" is enabled.
If laser does not move laser still be off, so focusing is possible only when laser is moving which does not make it a comfortable thing. This happens regardless using of M3 or M4.

Common procedure for focusing is:
- jog laser to a position when you put a scrap piece of material, then stop any movement
- send "S" command with a very low pwm power (to have the smallest - non burning - dot)
- send M4 to power on laser
- focus the lens
- send M5 to power down
- remove scrap piece and begin work

chamnit, 2016-12-22:
Yes. Both G0 and G1 without may be commanded without axis words. So you can set you laser on by `G1 M3 S1` without a motion. If it doesn't behave this way then it's a bug.

arkypita, 2016-12-22:
I have tested it right now and I can confirm that `G1 M3 S1 F1` power ON laser without movement.

chamnit, 2016-12-22:
Remember that G1 is modal. So once set any follow commands will be in the G1 motion mode. So M3 does not have to be on the same line. 

You can also set a default feed rate using a startup line. Like so `$N0=F500`.

PeterGrace, 2016-12-23:
Thank you @chamnit, I always learn something new when I chat with you!  passing a G1 without motion should do what I need just fine.