gnea/grbl #73: Frequently asking for Machine Status during GCode Cycle ?

nsiatras, 2016-12-24:
Hello everyone,

Maybe it is something that has to do with my electronics but I will ask.
Is it safe to frequently ask for machine status during the GCode cycle? 

I noticed that by decreasing the interval (~10ms) I send the '?' command during a GCode Cycle my machine makes a few Ghost moves and by ghost moves I mean that the machine makes moves that are not inside the GCode I am sending.

Thank you :)

My Setup:
Arduino Uno R3
Protoneer Shield
Nema17 Stepper motors and A4988 drivers

chamnit, 2016-12-24:
@nsiatras : 200ms-500ms is common. Not 10ms. For reference, it takes 115200 baud about 6ms to transmit a 70 character message.

closed 2016-12-24 by chamnit

nsiatras, 2016-12-24:
I wasn't very clear before.
I am waiting for the GRBL to respond and then after 10ms I send the '?' command.
So it is 10ms after the last received Status message.

109JB, 2016-12-25:
@nsiatras : What @chamnit is trying to say is that you are requesting status reports way too frequently.  From the 0.9 wiki it says this about status reports and should be no different in 1.1:

> We recommend querying Grbl for a ? real-time status report at no more than 5Hz. 10Hz may be possible, but at some point, there are diminishing returns and you are taxing Grbl's CPU more by asking it to generate and send a lot of position data.

I performed a test to see how long from a ? request is sent to when one is received and it is about 15ms at 115200 baud. Add your 10ms wait and you are basically requesting a status report every 25ms, which is 40 Hz, or 8 times more often than recommended.

In a nutshell, reduce the frequency of the status requests.

nsiatras, 2016-12-25:
That was very helpful. Thank you very much :)