gnea/grbl #76: Soft Reset Behaviour

Saublestarman, 2016-12-27:
I need to issue a feed hold randomly during operation and then send a different set of G-codes.  I am using  cntrl-X soft reset to flush the buffers but this also causes the machine to unexpectantly move a random amount.  I would have expected only the buffer flush.  Is this a bug? Versions 0.9 and 1.1 both exhibit this behavior.
I am developing a robot quilting machine and it needs to be stopped when the thread breaks and then reversed direction/ jogged to repair the thread and finally resume sewing.  Is there a better sequencing method I should be using?

chamnit, 2016-12-27:
@Saublestarman : A jump during a reset usually means that you need to set your step idle delay setting to 255 to keep them enabled during a reset.

Saublestarman, 2016-12-27:
I just gave your suggestion a try but the symptoms remain.  Just to be clear...when I issue a soft reset after issuing a feed hold not only does the machine move a random amount...the XYZ displayed positions also move.  It's as if during a soft reset the core positioning loops are allowed to move a little bit further along the tool path.

chamnit, 2016-12-27:
@Saublestarman : Can you post your $$ settings and $I build info string? Also describe a typical scenario that I can reproduce. It shouldn't move after a feed hold, but I'll take a look.

109JB, 2016-12-27:
@Saublestarman :  You are sending the "!" and the ctrl-x in too fast succession.  What you should do is send the "!" character and then wait for the machine state reported by grbl to go from "Hold:1" to "Hold:0" before sending the ctrl-x.

winder, 2016-12-27:
@109JB I was thinking that too, but I thought an alarm happeneds in that
case.

Saublestarman, 2016-12-28:
Right now I am testing my application in such a way that I manually send the feed hold.  Basically I set everything in motion and then manual click a gui button to send the ! feed hold.  I wait until everything stops and usually wait a few seconds so I can make some notes regarding positions etc.  Then, typically after 5 or more seconds of no motion, I send a Cntrl-X soft reset...again using a gui button that I press.  It's then that I notice the random amount of motion which always appears to be a small continuation of the tool path.  Lately I have switched to using Grbl Panel to further test this issue.  It is reproducible using Grbl Panel just using the Hold and then seconds later the Reset button while the machine is moving.
 
[image]

[image]

Saublestarman, 2016-12-28:
I am using the default build settings.  I have never tweeked them.

chamnit, 2016-12-28:
Did you ever try setting your step idle delay to 255? It's written with a $1=255 command.

Saublestarman, 2016-12-28:
Sonny,
I did try doing that.  I tried setting it to 255 and now I have set it back to 10.  My latest test was to leave the steppers unpowered and just exercise the Arduino UNO.  Running and then HOLD works fine…calculated positions look great even though nothing is physically moving.  Manually issuing a Cntrl-X soft reset during a feed hold immediately causes all three position values to move a small amount…always along the original tool path.

Peter

chamnit, 2016-12-28:
@Saublestarman : I'm not exactly sure what you are seeing and what would cause any motion after a reset. How much are you talking about? A tenth of a millimeter? A fraction of a step? How are you measuring this? Is it only in GrblPanel? Have you tried another GUI or directly connect to Grbl through a serial terminal? 

That said, you will want to keep your steppers always enabled upon a reset with $1=255. If its a value less than 255, it will disable the stepper drivers momentarily. This can cause some drivers to reset micro stepping position and shift slight when re-enabled. If you have some resistance in your axes or some stored energy making it want to move in a particular direction, that momentary disabling can cause the machine to move when the steppers are disabled.

109JB, 2016-12-28:
I just tried GRBL panel and could not duplicate this.  I also could not duplicate it in my own GUI.

Saublestarman, 2016-12-28:
This is very strange.  I just tested it again with Grbl Panel.  I used two separate Arduinos...one is loaded with Grbl 0.9j and the other is loaded with Grbl 1.1e.  Both behave similar.  I load a valid G-code file into Grbl Panel.  Then I tell it to run and after a few seconds I press Hold.  I note the XYZ positions and then press Reset.  The positions then move a significant number of millimeters further and then come to a stop.
[More GCODE testing.txt](https://github.com/gnea/grbl/files/676318/More.GCODE.testing.txt)

I included the G-code file I am using because I suspect that it may play a role.  My application starts with a design that is described with a series of XYZ co-ordinates.  I simply convert these to G1 codes and send them to the Grbl as listed in the attached file.

I have noticed that if I send a simple command like X1000 manually to Grbl and while it is executing I invoke a Hold and then Cntrl-X that there is no secondary motion.

Saublestarman, 2016-12-28:
My own application shows this issue so I decided to switch to a more refined application (Grbl Panel) to make comparisons.  This is not an electrical issue with the stepper motors.  The movements I am see are much much larger that a few steps.  Several millimeters typically which is hundreds of steps.  I am now doing all of my testing without the steppers attached because this issue is purely a software issue.

chamnit, 2016-12-28:
@Saublestarman : The only thing I notice is the size of the motions. They are unusually small. On the order of 20 microns (0.001") each. This translates to just a handful of steps per g-code block at 250 step/mm. This might have something to do with what you are seeing.

In your application, does altering this length to something more reasonable like 0.1mm or greater change this behavior?

jahnj0584, 2016-12-28:
I think a video may help. I just used UGS this break and when I stopped and
reset during a job there is no movement in my machine or the coords in the
DRO

109JB, 2016-12-29:
I think I have this figured out.  

I replicated this in a test program just now, and then replicated it in GRBL Panel with a different test program. If it is what I think it is then it is not a GRBL problem but a GUI problem.  

When using the send-response protocol a GUI is primed and ready to send the next line of the g-code and is simply waiting for an "ok" response from Grbl.  When you place it into hold, this doesn't change and the GUI is stilled primed and ready.  Then when you issue the reset command, if your GUI is set to do anything on startup then an "ok" is received and the send loop releases that last primed command from the run loop that it was waiting to send. Grbl itself doesn't send separate "ok" responses on reset, but appends them to the end of the lines such as for execution of the $N commands.  GRBL-Panel however sends several $ commands such as $G on reset which will prompt an "ok" response on a line by itself.

In my gui in the run loop there are several places in the run loop routine with checks to exit the loop.  Any reset button push triggers an immediate dump out of the run loop preventing the queued line from being sent.

Saublestarman, 2016-12-29:
I think that's the issue.  I just finished testing with Grbl Panel.  If I Hold and then Reset I see a random motion.  If I Hold and then Stop(black square) and finally press Reset the positions don't change.  I will probably find a similar bug in my application.  Thanks for sticking with this.

Regards to all

Peter

closed 2016-12-29 by Saublestarman

gerritv, 2016-12-29:
@109JB GrblPanel does have code to do this but apparently it is not working quite as expected :-( I will create an Issue for myself to look at this edge condition.
Thank you for your troubleshooting.

gerritv, 2017-01-17:
Fixed in GrblPanel 1.0.9.11