gnea/grbl #77: Just a curiosity about GRBL and ESP32S

Marco70cn, 2016-12-27:
Is there a chance to use GRBL on ESP32S?
If not compatible (as i suppose), will be difficult try to adapt?
I think use an ESP32S instead of Arduino should be 1000% better :-)

chamnit, 2016-12-27:
@Marco70cn : I haven't looked into ESP32's that much, so I can't say what the hardware limitations are. But, as far as I know, the compilers and toolchains are still not ready for general use, while Atmel and other major MCU manufacturers' toolchains have been around for a significantly longer time.

closed 2016-12-27 by chamnit

mkeyno, 2017-11-19:
@chamnit  how about now ? ESP32 is on the edge and FreeRTOS work very well now , it could easily assign  core for specific task,  it would be good idea to dedicate one core for GRBL core and another for user interface with nice webhost , even laserweb4 easily could uploaded on it