gnea/grbl #82: Problem with long tables

quangvi, 2017-01-05:
Dear @chamnit,
I'm having problems with long tables and high-speed machines, when I adjust the 3 axis = 1000 step/ mm.
(Reason Servo Driver is 5000 step/round , Ballscrews is 5mm/round.)
Reach almost 2 meters or more, Position began arising errors by the float variable is limited in floating point What do you think about this issue.? Should conversion from float to double functions and variable?

chamnit, 2017-01-05:
@quangvi : First, the AVR doesn't support double precision floating point values. Even if it did, it would an enormously costly variable to use on an 8-bit processor. Second, you shouldn't be seeing precision issues. Steps are tracked with int32 values, which have a max of 2 billion steps. Float values are accurate to 7.2 digits and should not encounter precision issue until 10-20 of meters of travel in your case. Third, you are likely exceeding the maximum rated step frequency of 30kHz. At 1000steps/mm, a  feed rate of 1800mm/min would be the max you could go before exceeding 30kHz. At this point, Grbl will begin to become unstable.

If this not what you are seeing, it would help if you posted your $$ settings and $I build info string. Also, explain in detail what you mean by position errors. How much and how it happens. The problem could be something entirely different from your original assumption of floating point being an issue.

quangvi, 2017-01-05:
thank @chamnit , my setting Grbl 1.1e ['$' for help]
$0=4
$1=255
$2=0
$3=0
$4=1
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=7
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=1000.000
$101=1000.000
$102=1000.000
$110=3000.000
$111=3000.000
$112=3000.000
$120=800.000
$121=800.000
$122=800.000
$130=8000.000
$131=8000.000
$132=8000.000
ok
[VER:1.1e.20161219:]
[OPT:M]
ok

quangvi, 2017-01-05:
i try a feed rate of 1800mm/min position no lost for 3 Axis ,but AC servo display Unstable velocity with acceleration 800mm/s^2 , it's ok about 200mm/s^2

closed 2017-01-05 by chamnit

quangvi, 2017-01-05:
I have find a problem and  solve this Problem thank @chamnit