gnea/grbl #83: Milling on non-plane Surface

oogre, 2017-01-06:
Hello, 
I wonder if someone here, has any ressources for me on this topic.
I'm working on a transportable CNC. It should be able to mill on approximately plane surfaces.
A sensor detects the distance between the surface and the milling tool.
GRBL has to update the Z-axis position on the fly with the value measured by the sensor.

chamnit, 2017-01-06:
@oogre : Typically machining processes don't do on-the-fly adjustments to the tool path. THis would be a specific application and would likely require custom software. However, there are tools to probe the surface prior to the job and will adjust the z-height tool path. But once running, it can't change. bCNC is the best tool for this, and you might be able to easily integrate an on-the-fly tool into it. It's written in Python and very well organized.

closed 2017-01-06 by chamnit

oogre, 2017-01-06:
Thank you for this info