gnea/grbl #84: Change version to numbers only

cprezzi, 2017-01-09:
Because of protocol changes between different versions, host softwares need to detect Grbl version to act accordingly.
It would be easier to check version "greater than" or "less than" if the version number would only have numbers, not letters.

chamnit, 2017-01-09:
@cprezzi :  I actually prefer the number+letter scheme over a long single numeric value. It's subjectively easier to read, shorter, and not misinterpret by human eyes. Since both the number and letter values always increase with updates, you could convert it to a numeric value internally to do the check.

chamnit, 2017-01-09:
@cprezzi : Also there is already another build value in the $I build info string. It's a push date in the form of year, month, and day. This value always increases and you can use this to determine which version is which.

closed 2017-01-09 by chamnit

109JB, 2017-01-09:
Also, strings can be compared using less than, greater than and equal to checks as well.  I have used this to alphabetize arrays of strings for another program I wrote a while back and it worked fine there. The only caveat is that upper and lower case are a bit swapped Since we are talking about version numbers, the format and case will be consistent, so just checking the version string should work anyway. At least I think it should. For example:

a is greater than A
a is less than b
z is greater than f
1.1e is less than 1.1f
1.1e is greater than 0.9j

electrokean, 2017-01-09:
Until you hit double digits of course :)

cprezzi, 2017-01-09:
Not usable as 1.11c would be less than 1.1e!

gerritv, 2017-01-09:
The protocol changed from version 0.9 to 1.x. No big deal to see that the first number after Grbl is 0 or 1. Why make this so complex? Trivial to code.

chamnit, 2017-01-09:
@cprezzi : True, but just compare the float value then the letter. And, do an toupper() or tolower() to the character to make it consistent. I make consider altering this scheme for the next ARM project, but it's not going to change for Grbl, as its direct development is nearing an end.

109JB, 2017-01-09:
> The protocol changed from version 0.9 to 1.x. No big deal to see that the first number after Grbl is 0 or 1. Why make this so complex? Trivial to code.

I agree.  The new GUI I am writing does just that

> Not usable as 1.11c would be less than 1.1e!

We have to get through however many 1.1a-z, 1.2a-z...1.9a-z versions are released before we get there.  I think that will be quite some time.

cprezzi, 2017-01-10:
I know how to parse that value, but it's just not convenient that all frontends need to do it for themselfs. A simple build number (int) would be much easier.

@gerritv It's not as easy as you think, as there was an security relevant change between 1.1d and 1.1e. (before 1.1e the laser was not stopped automatically on hold!)

chamnit, 2017-01-10:
@cprezzi : To be fair, Grbl v1.1d and prior are all beta releases to work out kinks and oversights. Grbl v1.1e is the final master release and there are no planned changes for this version. Just bug fixes.

cprezzi, 2017-01-10:
@chamnit : I know, but we cannot be sure that nobody will try with one of these outdated beta releases and I don't want to be responsible for a injury or fire. You know what I mean.
Thats why I immediately close the connection and report an error if such a version is connected.

Ok, I don't want to bother you anymore with my convenience thing. It's working like it is.