gnea/grbl #85: Reducing size for Nano

lberstein, 2017-01-09:
Hello everyone. This is my first time on github, and not very sure how everything works, so forgive me if I make mistakes :)

I'm trying to use grbl to run a coreXY plotter, using a Nano. Of course I inmediately run into problems as the code is just a tiny bit to big for the Nano (103% capacity). I read the documentation extensively and saw that this issue is already tackled in the wiki. However the solutions provided there are not an option as I don't have another board to do any bootloader cross-burning, nor skipping the bootloader altogether.

However, this is just a 2D plotter, meaning I don't have any need for spindle controls, Z axis, lasers, safety doors nor anything like that. Just 2 steppers for the CoreXY, and a servo for lifting/dropping the pen (I already found the code for adding servo support).

So my question is: what can and should I remove to make the code light enough to fit in the Nano? I know basic programming in some other languages and have been going through all the code but I don't want to delete something that could cause trouble.

ANY help will be greatly appreciated!

mayhem2408, 2017-01-09:
@lberstein I run GRBL 1.1 on my CoreXY laser engraver running a Nano. The Nano bootloader is large and slow. All I did was flashed the Uno bootloader onto the nano which is a little bit smaller and a lot faster. I can then write the firmware using the Uno option in the Arduino IDE and it has been working perfectly for months.

chamnit, 2017-01-11:
@lberstein : Unfortunately you will need an AVR programmer or a spare Arduino (that mimics one) to update your Nano's boot loader.

closed 2017-01-11 by chamnit

lberstein, 2017-01-11:
Yes, that's what I thought. So I got back to my original question, sorry for deviating from it:

What can I remove from the code to make it light enough to fit in the Nano? I know basic programming in some other languages and have been going through all the code but I don't want to delete something that could cause trouble.

As I said, I won't be using spindle control, PWM, safety door alarm, etc, but I'm not sure how everything interacts with everything else :(