gnea/grbl #86: Post update for Fusion 360, please vote

tklus, 2017-01-10:
Hey Guys,
I am trying to get the fusion 360 post guys to update the GRBL generic post to include laser function. the more votes we get the faster they will get it updated.

here is the link.

http://forums.autodesk.com/t5/hsm-ideas/update-grbl-post-to-allow-laser-program-posting/idi-p/6795471

wind-rider, 2017-01-12:
Are there any other changes that need to be included? Other gcode generation needed?

I read that there is new G28/G30 behavior, is there an update needed for this?

Strooom, 2017-01-29:
I have written a GRBL post processor for Fusion360 some months ago. I think it is better than the default included with Fusion.. Let me know what you need for your laser function and I'll try to include it. See my repositories for more info.

My post processor does not use G28 or G30 but a simple G53 to go home after the job, but again also that could be improved.

tklus, 2017-01-29:
That's great! One of the issues with laser programming in fusion is when you post the program, the post processor fails and complains that laser cutting is not supported. not sure if any other changes need to be made other than that.

There is a smoothie post processor that will output the code for laser but I have to go in and edit the code to make it work. I seem to remember having issues with it not posting the feed rates for laser moves. I would look at the smoothie post and see what they have done and add the laser code to your grbl post.

http://cam.autodesk.com/posts/post.php?name=smoothie

tklus, 2017-01-29:
here is the error I get when trying to post with your new openbuilds post
[image]

TheHandsomeDog, 2017-01-29:
Strooom. I see in your implementation of the fusion post processor that you have chosen to remove the M06 comands it is not supported. May I request that instead the post processor inserts a macro similar to those discussed here https://github.com/cncjs/cncjs/issues/118

Strooom, 2017-01-30:
@TheHandsomeDog I will take a look at it.
I removed M6 from the output, because GRBL doesn't support it, and if present it caused some GUI's to pause on it..
I'll take a look at expanding an M6 tool-change into a 'manual' tool-change macro. However I doubt that I want to use it myself, because on machines with timing belts (as mine), you often move the spindle while changing the milling bit, and so the machine needs to be homed again.. 

Can you explain why you prefer a single larger GCode file with 'manual' tool-changes, versus a few smaller files, each for 1 tool only ?

Strooom, 2017-01-30:
@tklus I'll take a look at that error. No idea why these 2D devices would not be supported as they use a subset of 3D toolpaths..

Strooom, 2017-02-01:
@tklus 
I now understand the cause of the error : each post-processor has a setting which kind of operations it supports. It was set to :

`capabilities = CAPABILITY_MILLING;`

I've changed it to :

`capabilities = CAPABILITY_MILLING | CAPABILITY_JET;`

So now when Fusion360 posts a ''WaterJet' operation, it's no longer throwing that error.
The post-processor was still outputting some header and footer GCode which contains Z axis commands.
I (conditionally) removed the Z-axis movements for Waterjet operations.

Give it a try and let me know how if it works.

tklus, 2017-02-01:
Is the updated post on your git hub?

tklus, 2017-02-01:
I haven't tried the post yet but am concerned about removing Z axis movement. there are times where I will make multiple passes in a material to cut deeper and I will set the Z to step down for each pass. with the z axis movement removed, will this still work?

tklus, 2017-02-01:
another feature that would be super helpful is the laser power settings on the post processor setting page. this shows up when you go to post. I have attached the smoothie post to show what I mean. this will give users with different laser settings to change what the max power output will be. I know some have max laser set to 1, some to 255 and some to 1000. this will set what the S___ speed will be in the Gcode.
[image]

Radiu, 2017-03-09:
Maybe of topic but can someone maybe give me a understanding  to figuer out this
i start machine and i start by doing a homing $H  
i jog to a location to the start point off my material lets say its in position X-100 Y-100 Z-10
when im, on the position i send command G54 to record my position by sending a  G10 P1 L20 X0 Y0 Z0
then i send home again $H
i load a file and start the job

now the problem is when using like inventors hsm when i open the file that contains the G code i see something like
`%
(1001)
(Tabuleiro_arduinos)
(T1  D=1.8 CR=0 - ZMIN=-7.5 - flat end mill)
G90 G94
G17
G21
G28 G91 Z0
G90

(2D Pocket1)
M9
T1 M6
Z10
G54
M9
G0 X8.099 Y3.306
S1000 M3
Z4
G1 Z-2.32 F1000
G19 G2 Y3.126 Z-2.5 J-0.18
G1 Y0.793`

i star the job and machine goes in the oposite direction hits switches and stops

I notice if i remove the line that contains G28 G91 Z0 then when i star job it will work ok and will go the the position recorded of G54, it seems that there is a relation with the G91 line that will increment

now the question is how can i make the postprocessor to dont generate the line with the G91
or there is something im missing here?

thanks

swarfer, 2017-03-10:
why do you home again?   homing the first time sets machine 0,0,0, then you define part 0,0,0 with the g10 command,    no need to home again!

you should be using Strooms GRBL post that does not send a G28, or otherwise turn it off.

vMeph, 2017-03-10:
Was doing  home again just to test it out sense i never used the G54 command..
Is the fusion 360 post processor compatible with Inventor?

X3msnake, 2017-03-10:
yes

neilferreri, 2017-03-13:
You could always use your g28. I set mine near the front left to make tool changes & jogging to part origin easier.