gnea/grbl #87: Getting Error:8 when runnon $N Cmds

hdrider465, 2017-01-12:
Getting the following error when trying to set $N0 and $N1

Cmd is $N0=M5G92X0Y350Z0F2500

Response is error:8

gerritv, 2017-01-12:
The error codes are all documented, easy to look up. See the doc directory in source code

hdrider465, 2017-01-12:
Yep - | **`8`** | Grbl '$' command cannot be used unless Grbl is IDLE. Ensures smooth operation during a job. |

Issue is that no matter what $N commands through Error 8
All other $ commands work

usbcnc, 2017-01-12:
@hdrider465 , can you post your settings?

chamnit, 2017-01-12:
@hdrider465 : Are you certain that Grbl is in an IDLE state?

hdrider465, 2017-01-12:
$0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000011)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=6 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$22=1 (homing cycle, bool)
$20=1 (soft limits, bool)
$21=0 (hard limits, bool)
$23=1 (homing dir invert mask:00000000)
$24=40.000 (homing feed, mm/min)
$25=2500.000 (homing seek, mm/min)
$26=25 (homing debounce, msec)
$27=1.00 (homing pull-off, mm)
$40=1 (laser mode, bool)
$100=200.000 (x, step/mm)
$101=200.000 (y, step/mm)
$102=200.000 (z, step/mm)
$110=2500.000 (x max rate, mm/min)
$111=2500.000 (y max rate, mm/min)
$112=50.000 (z max rate, mm/min)
$120=500.000 (x accel, mm/sec^2)
$121=500.000 (y accel, mm/sec^2)
$122=50.000 (z accel, mm/sec^2)
$130=481.000 (x max travel, mm)
$131=355.000 (y max travel, mm)
$132=100.000 (z max travel, mm

$N0=M5G92X0Y350Z0F2500 ( Error: 8)
$N1=F400S5 ( Error: 8 )

hdrider465, 2017-01-12:
It is Idle for any other $ cmd that send to it. Just not the $N cmds

chamnit, 2017-01-12:
@hdrider465 : What does Grbl's status report say? Idle or Alarm?

hdrider465, 2017-01-12:
<Idle|MPos:0.0000,0.0000,0.0000|Bf:15,127|FS:0.0,0|Ov:100,100,100>
ok
Done

In checking this though - I found that if I send $X and then the $N cmds they do get set.

usbcnc, 2017-01-12:
You have $22=1.

hdrider465, 2017-01-12:
Yes - Don't I need that for homing?

hdrider465, 2017-01-13:
Appears to be an issue with homing. 
In config.h I have set this ( I have an X Y Axis Laser - No Z )
// NOTE: Defaults are set for a traditional 3-axis CNC machine. Z-axis first to clear, followed by X & Y.
//#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))                // REQUIRED: First move Z to clear workspace.
//#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
// #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable
// NOTE: The following are two examples to setup homing for 2-axis machines.
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))  // NOT COMPATIBLE WITH COREXY: Homes both X-Y in one cycle.
// #define HOMING_CYCLE_0 (1<<X_AXIS)  // COREXY COMPATIBLE: First home X
// #define HOMING_CYCLE_1 (1<<Y_AXIS)  // COREXY COMPATIBLE: Then h
ome Y

But when I run $H it first tries to Home Z with a G1 Z5 and sets an alarm 
All subsequent moves give error 9 and $N settings give Error 8 until I send a $X

chamnit, 2017-01-13:
@hdrider465 : "Alarm" state is not an "Idle" state. Homing upon power up sets an alarm, as well as a failed homing cycle. Make sure you are editing the config.h file inside the Arduino library folder, not your downloaded folder. The import makes a copy there.

closed 2017-01-13 by chamnit