gnea/grbl #88: Low memory available, stability problems may occur.

lucky62, 2017-01-12:
Hello, 
I have a problem to upload the grbl into Arduino. No customization (default config) - just downloaded from github.

Arduino IDE shows the following message:

> Sketch uses 30,288 bytes (93%) of program storage space. Maximum is 32,256 bytes.
> Global variables use 1,631 bytes (79%) of dynamic memory, leaving 417 bytes for local variables. Maximum is 2,048 bytes.
> Low memory available, stability problems may occur.

Sketch is not uploaded into Arduino.
Please help.

Jan

------------------------------------------------
OS: Linux Mint 18.1
IDE: Arduino IDE 1.6.12
Board: RobotDyn UNO R3 (CH340/Atmega328PA)

chamnit, 2017-01-12:
@lucky62 : This is only a warning. Please read [this](https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#i-get-this-low-memory-available-stability-problems-may-occur-warning-after-i-compile-grbl-is-there-a-problem).

closed 2017-01-12 by chamnit

109JB, 2017-01-13:
> Sketch is not uploaded into Arduino.

So did it not upload.  The warnings wouldn't prevent this and the IDE should have more info if it truly didn't upload.

lucky62, 2017-01-13:
This was my mistake. Load is quite fast and no info message about successful upload. Later I found that grbl was successfully loaded - getting message in serial monitor.

Thanks for support.