gnea/grbl #91: Question: Tool Number

cheton, 2017-01-14:
Select Tool is not supported in Grbl, and it always displays T0 in a $G response. Just curious, why the line 924 in gcode.c said it only tracks tool value, but the line 313 never recorded the value with a T?

Tracking tool number might be useful for GUI to run corresponding macros for Tn if GUI supports tool changes.

https://github.com/gnea/grbl/blob/master/grbl/gcode.c#L924
```c
// [5. Select tool ]: NOT SUPPORTED. Only tracks tool value.
gc_state.tool = gc_block.values.t;
```

https://github.com/gnea/grbl/blob/master/grbl/gcode.c#L313
```c
case 'R': word_bit = WORD_R; gc_block.values.r = value; break;
case 'S': word_bit = WORD_S; gc_block.values.s = value; break;
case 'T': word_bit = WORD_T; break; // gc.values.t = int_value;
```

chamnit, 2017-01-14:
@cheton : That's a bug. Thanks for the report.

chamnit, 2017-01-14:
Fixed

closed 2017-01-14 by chamnit