gnea/grbl #93: Update

robcop2000, 2017-01-15:
I have tryed to update to GRBL 1.1e. 
My configuration is Arduino Uno- GAUPS-Stepper DRV 8825.
GRBL 1.1.e- bCNC 0.9.7 as controller
As soon as I have update I have tried to use jog mode. As soon as I have moved Z axis, the motor turn at hight speed and after stop. Now Z motor is not moving anymore.
As a try, I have changed, the motor, the stepper driver, and the arduino board, but the problem is still alive.
Any suggestion.
Thanks
Roberto

chamnit, 2017-01-15:
@robcop2000 : The upgrade to Grbl v1.1 erases your old EEPROM settings. You'll need to update your settings. And make sure the feed rate used for jogging is correct.

closed 2017-01-15 by chamnit

robcop2000, 2017-01-15:
I undestand what you say,but now for me the problem is to understand what has burned.
I sedt the correct value but the z asxis doesn't work.
- Is not the motor because i chenge it whit one new.
- is not the arduino because I flash another one and there is the problem
- Is not the stepper motor, because i change it.
- I'have tryed to install GRBL 0.9j, but now doesn't work.
Maybe is the GAUPS Arduino-compatible Stepper Shield, (I have only one of this) Should be greath if is this, understand which component, so I will try to chenge it.
Thanks
Roberto

vMeph, 2017-01-15:
Maybe the driver of the motor?

robcop2000, 2017-01-15:
I have already change
- Motor
- Arduino board
- Stepper motor driver
So I assume that can't be the

vMeph, 2017-01-15:
What happens if you put the motor z conected on driver x, and the motor x on the the driver z

vMeph, 2017-01-15:
Assuming that now the motor x will not work on driver z
Dont know that shield but Im assuming you have a4988 or drv 8825 did you ajust the pot voltage on the driver z correctly?

robcop2000, 2017-01-15:
Is not the first installation. The CNC have worked for more than one year. I didn't touch the Driver (are DRV8825). So I assume they were fine.
I have only made an update.
I will try to change  the connection of the motors to test them.

chamnit, 2017-01-15:
@robcop2000 : Do a process of elimination. Disconnect everything and test by only adding one thing at a time to figure what went wrong. Typically it's not Grbl and hardware related. Also, check your power supply if its maintaining voltage while running.