gnea/grbl #94: Feed Hold "!" releases Steppers

zeevy, 2017-01-15:
is this normal behavior?.
My machine having ball screw, z-axis sometimes moves down due to spindle weight is if i use "!" command.

zeevy, 2017-01-15:
This wont happen if i use the setting "$1=255"

jahnj0584, 2017-01-15:
Yes that intended. They shut off after a job or jog as well, you just
haven't noticed

zeevy, 2017-01-15:
Its ok, if steppers are released after the job or jog.
but its not ok, when they released in middle of the job where someone wantedly paused the machine.
one or more axis may be accidentally moved, if spindle is in air there are pretty good chances that it will move down there by loosing the position

jahnj0584, 2017-01-15:
so set $1=255 and this problem will be solved.

chamnit, 2017-01-15:
@zeevy : By default, Grbl disables the steppers after a motion. This was in response to users worried about overheating their drivers and motors early on in its development. It's not much of a concern now. Set $1=255 to keep them enabled. 

Just be aware that upon an alarm, Grbl will shut everything down, including the stepper motors. I don't intend to alter this behavior near term, so I would add some offloading springs or a motor brake for your z-axis when it depowers. BTW, using ball screws for a z-axis is somewhat unusual, because it requires power to keep it from dropping.

closed 2017-01-15 by chamnit

jahnj0584, 2017-01-15:
Ball screw Z's are very common, but they shouldnt be spinning a noticable
amount if the motor shuts off.

109JB, 2017-01-15:
> Ball screw Z's are very common, but they shouldnt be spinning a noticable
amount if the motor shuts off.

This is not necessarily true.  A Z-axis that has low friction can drift downward simply because of the weight of the head back-driving the screw.  Many commercial machines have a brake on the Z-axis screw so that this doesn't happen. Here is a [link](https://youtu.be/21LY6y5xf-Q) to a DIY CNC Mill that has this drifting problem.  Anytime the stepper is not powered the head can drift down. The solution during running is as mentioned, set $1=255.