gnea/grbl #95: Jog cancellation (0x85) sometimes not working

luben111, 2017-01-17:
Hi,
I'm using LabView to control GRBL v1.1 (downloaded the latest GRBL few days ago). Everything works like charm except that I spotted something which might be a glitch in the jog control.  The bug is that sending 0x85 to the GRBL sometimes is not aborting the movement. Let me explain in details the circumstances when this occurs. I recorded all data sent to the COM  port (char <…> is the 0x85) when the problem occurred :

G00 X-0.30                            (1)
$J=G90X-558F2000              (2)
…$J=G90X-558Y-2F2000      (3)
…$J=G90X-558F2000            (4)
…                                           (5)

1. X+ was just pressed,  send single step movement to absolute position

2. If X+ jog key pressed for longer than 500ms the program sends jog command and instead of sending frequently jog steps I'm targeting the end of the area (x= -558).   At this point the machine begins to move smoothly to the end of X axis (-558).

3. While moving in X direction I can also press the Y- jog key , 0x85 was sent before the command to abort the current jogging movements. This works well, now the head moves simultaneously to end of X and to the beginning of Y axis. 

4. Y- key released , send 0x85 to abort current jogging and send new jog commad targeting end of X axis (x=-558)

5. the X+ jog key released - sending 0x85 to abort the jog operations. The problem appears exactly here - in 2-5% of the cases the machine just ignores 0x85 command and continues to move. 

Let me know if you find something incorrect in my jog commands which may cause the problem.

In short - sometimes 0x85 command is not aborting the currently executed jog movements.

Best regards,
Luben

chamnit, 2017-01-29:
@luben111 : I'll look into this when I get the time, but first could you verify that Grbl is receiving the 0x85 command when you notice it fails? Not exactly sure what would be easiest way to do this. Maybe edit the serial RX ISR code to toggle a pin whenever it reads a 0x85.

luben111, 2017-01-29:
Good point, I'll check this

shamsiqbal, 2017-02-06:
@luben111  : you have mentioned that you are using labview would you like to share your work with me so i will be able to help/verify more/new features of grbl basically i am porting grbl to xmega avr where i am aimed to add G33 (tapping /threading ) g-code support and i need labview type application to test function integrity

luben111, 2017-02-07:
@shamsiqbal 
There is LabView VI which is handling the communication over the serial COM in LV environment  and in your case it doesn't bring any advantages compared to using any Serial COM terminal. This LV code is very sophisticated (state machine, position polling, etc.) and if you're not well experienced in LV it will be difficult to make it working. What you want to test for the G33 command and why you need this to be done exactly in LabView?

shamsiqbal, 2017-02-07:
@luben111 
i had worked in labview in past to develop many application for instrument interfaces and i can work on your vi too with little or no help from you 
here are some key things which can be done in labview 
pid tuning interface can be developed in labview (in your existing VI ) which is must thing for spindle closed loop control and it will took me lesser time  with compere to develop brand new GUI i have added closed loop pid control in xmega port 
after that i can integrate testing VI in existing VI for G33 function integrity/validity 
its all about that i can work in labview and its easy for me instead starting from zero 

i have retrofitted beaver vc5 cnc milling  machine it was a mix of xmega and arduino nano but next version i am going to use only xmega128a1u
just finished porting and adding xmega hardware driver support  now started working on jog keys that how to implement that function in grbl using GPIO of xmega i have added new grbl parameters for spindle pid tuning which need to be added in  grbl interfacing GUI i will also alter spindle and feedrate override with potentiometer attached with xmega adc channel and show the actual override % on GUI

luben111, 2017-02-08:
@shamsiqbal 
It sounds reasonable to use LV in your case. It will take me a little time to clean the code because it's a part from a large project. Let me know how I can send the files over email.

Porting to XMega128 sounds like a good idea (despite that very soon will be available port to ARM), do you have some PCB for it?

About jogging - the new GRBL v.1.1 has built in jogging control and I can say that it works really well - with the PC keyboard I can get technically the same behaviour as if using a pendant .

shamsiqbal, 2017-02-08:
@luben111 
well arm port doesn't solve any problems >-----------> will it gonna have G33 g-code support 
about arm port i knew it since v0.9 its no use at all for me 
if there is going to support for S-curve motion planning in grbl i can say 16 MIPS processing power is sufficient to control 6 axis closed loop dc servo control with virtually no speed limit on motion control thing its gonna be only limited by servo motor speed 
 i choose xmega because it have  three hardware quadrature encoder counting option have double the processing power of avr mega and it can be used generate more steps then mega avr 

about xmega pcb i have built breakout board using photosensitive film method you can build one using toner transfer method its a simple smd to dip breakout board once i am done with porting i will build complete pcb using my own built cnc milling machine 

you can email me files on shams@bosonelec.com 

i am using v1.0 not v1.1 for grbl port because i have found some issues in v1.1 where my newly added parameters wont get saved properly 
     
    @chamnit : good job to getting juices from arduino uno but plz admit this thing  that its almost end of arduino uno available resources due to flash memory limit so plz dont make grbl mega difficult where adding new function will become difficult i dont want to add a new issue stated above because i can understand that i wont be getting any helping answer from you 
i think its not important to port grbl on ARM but to go for S-curve motion planing . you have chosen lpc17xx arm port but it will be good choice if you had chosen stm32 nucleo board due economical price and on board debugging option available on it plz have look on stm32 nucloe boards available from st. cortex m3 vs cortex m4 and fpu is big difference i have 3k kintes  mkv3 series cortex m4 mcu's in my stock but i still prefer stm32f4 family where standalone grbl based cnc controller is possible on stm32f429i discovery board in my opinion i will suggest you to plz consider stm32 family boards for grbl arm port

chamnit, 2017-02-08:
@shamsiqbal : Before you go and start saying things that you have no clue about, please try reading the other issues threads. In particular, the thread about moving on to ARM. There is a lengthy conversation there about this topic. Also the LPC port was done by a user and is not my primary choice, as you'll see if had you read the topic. And for everyone that is demanding to use this ARM chip or that ARM chip or even FPGA, I'm done talking about this. It's getting ridiculous.

tbfleming, 2017-02-08:
@shamsiqbal : my brother and I ported grbl to the lpc17xx because there's a very large number of CNC boards sold which use it, but running a different firmware. As far as I know, grbl-lpc has faster step rates and block rates than any other grbl port. Will M4 perform better? Maybe. We'll see when you or someone else port to it.

shamsiqbal, 2017-02-08:
@chamnit : 
it looks like that if i open issue then you will be helping me on issue 
 "**i am using v1.0 not v1.1 for grbl port because i have found some issues in v1.1 where my newly added parameters wont get saved properly**"
 
further these are your comments on [https://github.com/gnea/grbl-Mega/issues/17](url) thread 
**Grbl is being ported to 32-bit ARM**
never mentioned that its your port or someone else port i do remember that there are similar comment from you on different issue thread that you are busy with arm port but i still apologies for my comments about arm port 

in this thread i have pointed three things    
i) G33 (tapping/threading) g-code support 
ii) remain it compatible with other older version for everyone and add new function with minimal change for every user / developer v1.0 to v1.1 total revise for every GUI developer  
iii) S-CURVE motion planning 
dear   @chamnit : 
plz let me know if these were out of the scope of grbl 
for G33 i have started work on it 
for remain it compatible i have also faced issues 
for S-curve motion planing i will be working on it after complete port of xmega  
 
i have been reading grbl code since v0.8 i have less experience on grbl then you have and if i am not able to perform any additional  function its obvious thing to ask help from experts its not how things works but its all about code compatibility with existing function in grbl 

@tbfleming 👍 
good job and great approach to help community i do have three lpc17xx embed boards but still i am not going to use these boards you have mentioned speed performance its obvious thing that 16Mhz vs 100Mz thats the speed difference in mcu you are using but here is my main question do you think that a common stepper motor and stepper motor driver going to to accept high step rate and going to perform well 
 higher step rate and rpm poor torque curve for stepper motors and loss of actual position 
when i am done with xmega port i will going to upload a video on youtube that how my system works

chamnit, 2017-02-08:
> never mentioned that its your port or someone else port

Please stop contradicting yourself. You stated **this** in your last post. 

>  you have chosen lpc17xx arm port but it will be good choice if you had chosen stm32 nucleo board due economical price and on board debugging option available on it plz have look on stm32 nucloe boards available from st. 

Please understand that answering questions on these issues threads takes time and energy. It's for determining if there are bugs or discussing important issues. 

What is not for is teaching every random person how to code for Grbl. The code is well-commented for this reason. Nor does it make any sense to try to help you on an older version of Grbl, when everything is moving forward with v1.1. What you want would be a complete waste of time for myself and others here.

Also, all of the improvements you have suggested has been previously discussed in numerous locations and are already planned. Please try using Google and do some homework before posting here. It'll save everyone the precious time that they volunteer to help out. Time that should be spent on development or making something.

tbfleming, 2017-02-08:
> do you think that a common stepper motor and stepper motor driver going to to accept high step rate and going to perform well higher step rate and rpm poor torque curve for stepper motors and loss of actual position

ClearPath servos.

tbfleming, 2017-02-08:
Also, steppers on current-generation hobby lasers run close to 100 kHz. Future hobby lasers may need higher rates.

luben111, 2017-02-10:
About jog cancellation problem (I'm sorry for the thread contaminated by other posts):

In grbl.c (lines 91..) add the following code:

  // Determine if the line is a jogging motion or a normal g-code block.
  if (line[0] == '$') { // NOTE: `$J=` already parsed when passed to this function.
    // Set G1 and G94 enforced modes to ensure accurate error checks.

    **// 10/02/2017 modification done to detect if the JOG command was accepted. Scope should be attached to pin 4 of ISP header (MOSI - PORTB_Bit3)
    // check with scope do you get the pulses when the jog command failed (was not accepted)
    SPINDLE_ENABLE_PORT ^= (1<< 3);  // toggle pin 11 on Uno when new jog command was accepted**
    gc_parser_flags |= GC_PARSER_JOG_MOTION;
    gc_block.modal.motion = MOTION_MODE_LINEAR;
    gc_block.modal.feed_rate = FEED_RATE_MODE_UNITS_PER_MIN;
    #ifdef USE_LINE_NUMBERS
      gc_block.values.n = JOG_LINE_NUMBER; // Initialize default line number reported during jog.
    #endif
  }

It changes the state of PB3 (pin 4 of ISP header) each time a new jog command accepted. What I see is that each time I send a jog command it was accepted correctly. The sequence I used is similar to the sequence listed initially:
$J=G90X-558F2000 (2)
$J=G90X-558Y-2F2000 (3)
$J=G90X-558F2000 (4)
the scope snapshot is below - you can see exactly 3 changes of the signal:
[image]

So receiving the jog command is correct, what should be wrong is the cancellation of the running jogging. Now will add some another pin change to show the jog cancellations in time as second trace on scope.

luben111, 2017-02-10:
I made the following modifications into serial.c (line 161) in order to show the 0x85 cancellation in action - each time 0x85 is cancelling something PB5 will toddle (tisible on pin 3 of ISP).

@------------------------------------------------------------------------------------------------
            if (sys.state & STATE_JOG) { // Block all other states from invoking motion cancel.
              system_set_exec_state_flag(EXEC_MOTION_CANCEL);
              SPINDLE_ENABLE_PORT ^= (1<< 5);  // toggle PB5 pin (pin 1 on ISP)
            }
@------------------------------------------------------------------------------------------------

Here are the timings from the scope:
[image]
[image]

It seems there is really some problem and it occurs in the following conditions
1. Send a jog command
2. Send 0x85 command (to cancel the effect of previous command 1)
3. Send jog command  
4. Short time after you send (3) send new 0x85 to cancel all jog commands.

The misbehaviour is - if 0x85 was sent before command 1 was stopped  command 2 will be executed, i.e. 0x85 is not affecting jog commands which were accepted and waiting to be executed while previous command was cancelled.

My explanation is:
- After you send 0x85 (2) the machine begins to decelerate - cancellation of (1)
- you send new jog command (3) immediately after (2)
- if you send new command 0x85 (4) before command (1) was completed (head stopped)  it will be accepted  but command (3) will not be cancelled (it's hanging probably in some other buffer).

chamnit, 2017-02-10:
@luben111 : Thanks for digging into this. It's possible there may be some bug in the state machine. I didn't vet it as much as I have in the past prior to releasing v1.1. But, first: 
- With sending (3) immediately after (2), what exactly does Grbl do? Does it start to slow down and then accelerate again? Does it stop and then go? Or does it continue moving like nothing has happened?
- If it starts slowing down and accelerates again, It's possible that Grbl is doing what it is supposed to by stopping the machine. However, when you send the next command, it may have gotten queued somewhere between the serial buffer and the planner and executed immediately after the cancel completes.
- Regarding (4), sending another 0x85 jog cancel will be ignored, if the current one hasn't been executed and cleared yet. 

I suppose the question would be, what is the situation that you are sending a jog cancel and then still are sending jog commands? Does Grbl operate ok if you stop sending jog commands until it stops and then resume streaming jog commands? 

If so, perhaps I need to add a `[Jog End]` message for GUIs to know when the jog motion ends.

luben111, 2017-02-10:
@chamnit 
With sending (3) immediately after (2) - I can't distinguish what's exactly happening, for sure after sending (2) the machine begins to decelerate (there is a hiccup, not sure if it stops completely). It will be very difficult to get this data without some debugging tool. 

>>I suppose the question would be, what is the situation that you are sending a jog cancel and then still are sending jog commands? 

The situation is very simple and this is how my implementation of jogging is working on LabView:

1. In order to avoid constant sending of jog commands to the serial port I send single jog command which is targeting the end of the X or Y area. For example moving in X direction against one end (let's say it's X=300.00) I could send simply $J=-X299F2000 and everything will be all right. With this approach I get two benefits:
  a) I'm sending only a single command to the jog control (no need to resend commands)
  b) it's guaranteed that the jog will not exceed the working area of the machine (no need to check working coordinates)

2. If the  jog button was released - I just send 0x85 and movement is cancelled (this is working well)

3. The problem appeas if you use for jogging two or more buttons. If I press second button while the fist was kept pressed (to move simultaneously on XY) I send 0x85 to cancel the previous command and then I send the new command - for example $J=-X299Y-199F2000 . The machine makes a small hiccup and continues to the new positions, now moving on X and Y simultaneously.

4. The problematic moment is when the keys were released - when I release the two pressed buttons normally there is some delay between two events (releasing the first and releasing the second button). In this time gap the bad events occur - for example when I release the Y  button first I send 0x85 to cancel the movement on XY ( $J=-X299Y-199F2000) and then I send a new command $J=-X299F2000 to continue moving on X. On this stage everything works fine.

5. A short time after (4) the second button was also released and I send a new 0x85 to cancel all movements (this is the end of jog movement). And here the misbehaviour occurs - this 0x85 somehow doesn't clear the last jog command in the buffer.

Maybe adding [Jog End] message will not improve things because now we have to hang around the serial to see if it's accepting or not the 0x85 commands and if not to resend it. The best approach is if we can locate where this happen and to fix the logic.

chamnit, 2017-02-10:
@luben111 : 
- Regarding seeing if Grbl stops, you can send a status report when the jog cancel completes in protocol.c's cycle stop code. This should show you the current velocity, which should be zero. 
- Grbl's jog cancel was designed to expect the cancel to complete before send new commands. I was not expecting more commands to be sent during this time. Jog cancel is there to cancel long motions, not very quick ones. I can make this more robust, but this is how it was intended to work.
- In your case, send only a short time segment of motions, as outlined in the jogging document. Keep the planner just full enough to attain the programmed speed. Anytime there is a direction change, just send the new direction commands and let what's in the queue complete. Don't send a jog cancel. If done right, it'll actually run better this way because the planner can account for small direction changes, decelerate when it needs to, and keep the speed optimal.

luben111, 2017-02-10:
@chamnit 
- sending status report and getting the data on PC is very slow process - on USB it may take up to 30ms to complete, i.e. using the status report to take quick actions is not the best approach for PC programs (but it might be OK for controllers which communicate directly with the Atmega328)

>> Grbl's jog cancel was designed to expect the cancel to complete before send new command
Is there some event to show that jog cancellation was completed? 

If this was the way how the jog cancellation was designed I can make a patch and delay the 0x85 command to the GRBL so it will take effect. Another option is to send it several times by hoping that one of the commands will come in the right time. 

- I agree that sending short segments will make the jog movement extremely smooth (no hiccups) but on the price of excessive communication

chamnit, 2017-02-10:
@luben111 : 
- I meant to compile in a status report call directly in the source code. Not from a PC. So it generates one automatically when it reaches that point. Similar to how you set the pins.

- It's best to fix the jog cancel issue so that it doesn't miss a cancel command, rather than send multiple cancels. I'll probably take a look this weekend, but can't guarantee anything. My workload is full right now.

- Yes, sending short segments will have a lot of comm traffic, but Grbl can handle it. If there is a stream interruption, Grbl will just slow down to a stop until there are more motions.

luben111, 2017-02-10:
@chamnit 
>>It's best to fix the jog cancel issue so that it doesn't miss a cancel command

yes, this is the best approach - keeping some misbehaviour in the code and explaining to everybody how it should be avoided is not great. There is something else - very often behind a bug stays some other problem which is invisible.

BTW, the GRBL code is really a masterpiece, I really appreciate all your efforts to support such sophisticated and well working engine.

terjeio, 2017-03-01:
I have a similar problem but then I am trying to execute jogging by calling code directly, not via serial comm, so I am possibly on very thin ice. IMO the problem could be related to the EXEC_MOTION_CANCEL flag beeing reset before the cancel is completed, if a new cancel command is received before completion it will set EXEC_MOTION_CANCEL flag again and thus possibly cause all kinds of problems later? I am going to try to avoid this problem by not allowing the EXEC_MOTION_CANCEL flag to be set until the cancel is completed, not sure if this could work:

`	          case CMD_JOG_CANCEL:
	            if ((sys.state & STATE_JOG) && !(sys.suspend & SUSPEND_JOG_CANCEL)) { // Block all other states from invoking motion cancel.
	              system_set_exec_state_flag(EXEC_MOTION_CANCEL);
	            }
	            break;`

If not I will have to dig deeper...

luben111, 2017-03-02:
@terjeio The movement cancellation is actually working - whenever you send jog cancellation it will always stop the head. What is not working is purging the buffers of new incoming jog commands. So the scenario when you'll get jog misbehaviour is when you send the following sequence in short time:
1. send jog command
2. send jog cancellation
3. send new jog command
4. send jog cancellation
You'll find very often that 3 is not cancelled but 1 is always cancelled.

In my case instead of sending multiple small segment jog movements I'm sending single jog command targeting the end of the working area (left, right, top or bottom). When doing horizontal and vertical movement simultaneously (pressed for example left plus bottom arrows of keyboard) and when releasing both buttons almost simultaneously the described above sequence may occur. For example if moving left + up and releasing buttons I may get the following picture:
- head is moving left + up (there is active jog command)
- released up key:
     a) sending jog cancellation
     b) sending new jog movement to the left side
- released left key 
    c) send jog cancellation. NOTICE - this cancellation is not affecting the b) command

terjeio, 2017-03-02:
@luben111 : "purging the buffers" is the clue, since jog cancellation is picked off the input stream in real time the jog command to be cancelled may still (partly) be in the serial input buffer? I''ll check what happens if I flush the buffer on a jog cancel but I suspect it will not be a 100% foolproof solution.

terjeio, 2017-03-02:
@luben111 : Thinking about this some more before I commence testing: flushing the serial buffer, inserting a CAN (ctrl-X) **into the buffer** and use that to flush the line buffer in protocol.c may be even better.

terjeio, 2017-03-02:
@luben111 : my solution above seem to work, I think you should try it. There may be a edge case related to the time window from the job command is queued for execution to sys.state = STATE_JOG; is set (in jog.c), if the cancel arrives in that window it will be missed. This is what I did:

protocol.c

```
   while((c = serial_read()) != SERIAL_NO_DATA) {

      if(c == CMD_RESET)
    	  char_counter = 0;

      else if ((c == '\n') || (c == '\r')) { // End of line reached
```

serial.c

```
case CMD_JOG_CANCEL:
     serial_cancel_read_buffer();
     if (sys.state & STATE_JOG) { // Block all other states from invoking motion cancel.
        system_set_exec_state_flag(EXEC_MOTION_CANCEL);
    }
```

added (NB! ARM code in my port)
```
void serial_cancel_read_buffer(void) {

	rx_head = 1;
	rx_tail = 0;

	rxbuf[0] = CMD_RESET;

	GPIOPinWrite(RTS_PORT, RTS_PIN, rts_state = 0);
}
```
Disclaimer: I am still trying to get to grips with the innards of Grbl so bear over with me if I am messing things up.

luben111, 2017-03-02:
@terjeio  - where are these vars:
   rx_head
   rx_tail

luben111, 2017-03-02:
@terjeio 
Adding only the code
      if(c == CMD_RESET)
    	  char_counter = 0;
into protocol.c doesn't fix the problem

terjeio, 2017-03-02:
For the original code I guess it would be something like this:

```
void serial_cancel_read_buffer()
{
  serial_rx_buffer_tail = 0;
  serial_rx_buffer_head =1;
  serial_rx_buffer[0] = CMD_RESET;
}
```
Adding the code above does nothing, you need to have the serial_cancel_read_buffer in place also.

luben111, 2017-03-02:
@terjeio 
I added the code as you suggested. There is substantial improvement - the problem is very difficult to be reproduced now. From ~100 attempts I got only once the misbehaviour (where before I was able to get almost every second jog to misbehave). It's so rare that I did the experiments again to ensure it's not some glitch, unfortunately if you're persistent and if you try to release the keys with some specific timing you can see it again.

The main challenge is that we need to cancel the current movement but we can't just stop the head, it will take some time to complete the task and during this time window something bad can happen. It's something about buffers / queues.

terjeio, 2017-03-02:
@luben111 : great, and it confirms what I suspected - there is still a small time window where the cancel get lost. Could it be solved by moving the sys.state = STATE_JOG; statement a bit? Or by disabling interrupts for a little while in the correct place? I don't know, maybe @chamnit does? I will have a go at this later when I test the changes in my machine - I want jogging to be foolproof.

terjeio, 2017-03-03:
@luben111 : can you try this:

```
case CMD_JOG_CANCEL:
     serial_cancel_read_buffer();
     if ((sys.state & STATE_JOG) || sys.state == STATE_IDLE) { // Block all other states from invoking motion cancel.

        system_set_exec_state_flag(EXEC_MOTION_CANCEL);
    }
```
there is a chance it could solve the problem (or make Grbl fall over?).

If not a new flag (or a counter?) could be a solution: jog_in_pipeline - to be set early and cleared in the code above, It should then be checked for and cleared where sys.state is set to STATE_JOG. Marking sys.state volatile _could_ be required too since the code above is called when the processor is in interrupt mode - but then maybe not since it is in a struct.

terjeio, 2017-03-04:
@luben111  : when jogging fails for you does Grbl end up in suspended mode?

I finally got around to add keyboard jogging to my gcode-sender and when jogging fails Grbl is suspended, seemingly due to EXEC_MOTION_CANCEL beeing processed when state is STATE_IDLE. Somehow the state has changed from STATE_JOG when EXEC_MOTION_CANCEL was set. Sadly, the programming style does not promote setting breakpoints - I think I will change that so I can try to trace what is going on.

By the way, the "hardware" jogging I have wired to the controller seems to be a lot more robust, yet not perfect, I have not managed to get Grbl enter suspended mode when using that. This could be due to the way I have implemented command key processing, different time delays between key down and key up events? - most likely a lot shorter than via serial comm from the PC.

chamnit, 2017-03-04:
@terjeio : FYI, Grbl's state machine is currently being re-written to be more testable. The current codebase is the result of minimizing the compiled size. Some less savory things had to be done.

luben111, 2017-03-05:
@terjeio Thank you - I'll check this code change on Monday and will let you know how it behaves.

You mentioned that you did GRBL port on ARM - is it available somewhere to try it and is in mature state? I made a handwriting robot [[http://www.thingiverse.com/thing:1808184|Robot]] but I get lack of speed with Arduino UNO and I need to increase the pulse frequency 3 to 4 times.

terjeio, 2017-03-05:
@luben111 : I will look into publishing my port on Github, I have to learn how to use Git first. I keep my code in a Subversion repository, I _can_ give personal read access to that on a case by case basis.

"Mature" is not a term I want to use yet - "proof of concept" is more appropriate? This because I have integrated Grbl into the codebase I made for my CO2 laser engraver. The base mode of this is raster rendering much like a how a (needle) printer works, for this I have implemented my own stepper driver code. This allows me to control the laser precisely, eg. I can adjust laser power per pixel - useful for 3D engraving. I use a DAC for setting power, not PWM - this means the controller card I have made can be used for high speed lasers as well. A (cheap chinese) C02-laser tube is a slow beast compared to laser diodes...

I added Grbl to be able to do cutting and still have access to functionality provided by the base mode code, such hardwired keypad (for jogging etc), display and PPI mode laser control. As it stands now I believe I am very close to getting ready for putting my laser back in a working condition. Initially I used Mach3 for cutting, but I missed the lack of integration I now have available.

For porting Grbl I started with a Texas Instrument Tiva C LaunchPad (TM4C123GH6PM), I can port back my latest driver (and strip out all code in the driver related to the CO2-laser) if required. TI recently made their development suite (CCS) free of charge so it is (IMO) now a viable platform for personal use. CCS has a debugger which for me is a must...

I have a couple of surplus controller cards I am willing to give away for postage, no components and SMD soldering required for assembly.

For a bit more info about my laser see [http://www.buildlog.net/buildlog/view_log.php?id=2625](url)

luben111, 2017-03-05:
@terjeio I took a look at your page - you achieved amazing things. really impressed.

How many axis you have in the implementation? 

What performance you achieved (steps / s) ?

Do you have a breakout board to connect the steppers / sensors to your controller?

tbfleming, 2017-03-05:
@luben111 there's a port to the LPC 1769 here: https://github.com/gnea/grbl-LPC . I get up to 2000 blocks/s at 100 kHz step rate or 1000 blocks/s at 200 kHz step rate (needs a different set of define values to reach this).

luben111, 2017-03-05:
@tbfleming Thank you - 200 KHz with 1000 block/s is more than enough for what I need. Do you use serial interface for communications and what is the max speed?

One question - what is target controller for LTC GRBL? What is the best source for them?

tbfleming, 2017-03-05:
@luben111 it uses USB. It runs on Smoothieboard and Cohesion3D boards.

terjeio, 2017-03-06:
@luben111 : Thanks.

My card has support for 3 1/2 axes, the 1/2 is from beeing able to switch the Y-axis between two motors - I am considering adding a rotating axis for engraving on round objects.

I have not checked for performance and I am unsure how to measure it, when compiled optimized for speed the time spent in the stepper ISR is 1.5 - 4µS,  time spent in st_prep_buffer varies between 18 - 250µs, this when running a gcode program consisting of many small moves (in laser mode). Does this translate to approx. 4K/blocks per second? Anyway, is there a standard way of measuring performance that you know of? Not that I am too concerned about this, my main goal is to add gcode handling to my machine and I believe Grbl is more than fast enough for that.

A small breakout board is used for managing the wiring. The eStop switch and mechanical limit switches are wired in series in NC configuration and directly connected to the MCU. I have separate optical homing sensors, their output transistors are again directly connected to the MCU as the GPIO pins are schmitt-triggered.
I have no issues related noise, even when driving the tube in PPI mode (pulsing 15 kV at up to 15 mA).

DM420A stepper drivers run the motors, they have internal optocouplers so I drive them buffered via 74LCX157 as the GPIO outputs are limited in their current handling.

For rendering images with my laser I can choose how I want it done. I have not (yet) added any image processing capabilities to the GUI interface, I use Photoshop for preprocessing if needed. Grayscale rendering (128 levels) is supported and power profiles can be loaded in order linearise the output as (cheap chinese?) CO2 lasers are inherenly non-linear. I really want a Synrad laser, they are a lot faster than their chinese counterparts - but a lot more expensive :-( 

Yesterday I tested jogging via the front panel keys, this inputs jogging commands via a function I have added to Grbl, and so far I have not experienced any hiccups.

shamsiqbal, 2017-03-06:
@terjeio 
hi 
i have tested latency of stepper isr routine. Do not consider it under 4 Usec its almost bigger than 12 Usec 
best way is to simulate it in labcenter proteus with virtual com-port pairing and with different optimization level in atmel studio 
doing so will give you quick debugging and functional testing options 
i hope this will help you

terjeio, 2017-03-06:
@shamsiqbal : I am running Grbl on a 80MHz 32bit ARM M4F (with floating point unit - FPU). The timings I posted above are what I observed on my scope, the shorter ones seemed to be the most common. I would expect this processor to perform at least four times faster than a 20 MHz Atmel 8-bit processor.

The way I have implemented the hardware driver adds a bit of overhead compared to the original version, this will negatively affect the perfomance of my port. However, I consider it a small price to pay for the flexibility I have gained. The remaining issues I have are relatively straightforward to solve, perhaps the most confusing for me at this point is that limit switches are used for homing, perhaps due to lack of GPIO pins on the Atmel...

luben111, 2017-03-08:
@terjeio I made the changes and I did the tests but unfortunately I can still see (extremely rarely) the problem to occur. For sure it's much better, you can tell the difference immediately, but the problem is there. In my opinion cleaning the buffers is not that simple - jog cancelling command should not clear the buffers BEFORE the existing JOG command was not completely stopped and it should not clear the whole buffer. I think the core of the problem is that  jog cancellation command clears immediately the buffers and it doesn't take in account the order of commands (some commands may occur after the jog cancellation and they should not be cleared), i.e. jog cancellation command is exception and it executes immediately. The correct approach should be:
1.  Jog cancellation command should stay in the buffer as all other commands, for sure we need a flag/index showing  jog cancellation command presence.
2.  Once jog cancellation command was detected in the buffer if JOG movement is active the program should decelerate the head 
3. Once the movement was stopped completely the buffer will be cleared - delete all records from beginning UP TO the cancellation command. Here is the difference - if some other commands came after the jog cancellation they should stay in the buffer untouched! 

In short - if the jog cancellation command was detected we should:

a) if active jog movement is active - begin the deceleration to stop the head.

b) Once the head was completely stopped - clear all commands before the jog cancellation command + the command itself. But we should not affect commands after the command (if any).

terjeio, 2017-03-08:
@luben111 : this is not easy...

A jog cancellation command (0x85) should be acted upon immediately, never be buffered. The action should be to flush all (jogging) commands in the pipeline (in front of the cancel command) and cause the head to decelerate to a stop if a jogging motion is currently active (in the execution stage). Simple as that (IMO).

The code I added to flush the input buffers (serial and line) ensures that this part of the pipeline is taken care of, however it requires the sending program to properly stream gcode - by not sending a cancel in the middle of other (gcode) commands. Neither should jog cancel commands be sent when non-jogging commands are beeing processed. This because serial_cancel_read_buffer() _has_ to be called regardless of sys.state (or another flag has to be added to the state machine as suggested, I am not going to try that - better to wait for the proper ARM release then).

The tricky part is to prevent jogging commands in the remaining stages of the pipeline to reach execution and to properly start deceleration. This is where the state machine(s?) comes into play - the latest changes I made is for handling some cases where (IMO) it went wrong. It could well be it goes wrong in other places as well - but so far I have not observed that in my ARM port. What I did was to attach breakpoints in the code and watch how the code behaved, this obviously requires a debugger and an idea of how the state machine is supposed to work. What is a bit problematic is that code affecting the state machine is scattered all over and hard to follow - at least for an old fart as I am.

Currently I can only see one potential issue with the code related to the problem, sys.state is checked when in an interrupt handler (case CMD_JOG_CANCEL: ...) - I am unsure if this requires the sys.state to be marked as volatile, my programming skills are not good enough to determine that. Maybe this could be a reason why my ARM port seems to be working perfectly now? Different compilers and different MCUs to boot? Or could it be due to the fact that I use hardware handshake for the serial port (RTS/CTS)? And I have a 1024 bytes RX buffer with a high water mark for handshake at 900 bytes.

Since I am writing my own gcode streamer I checked it by connecting it to a terminal (via virtual serial ports). This to be 100% sure that the output generated for jogging was proper - no missing cancel commands and no cancel commands where they should not be. Windows programming can be tricky too...

luben111, 2017-03-08:
@terjeio 

>>A jog cancellation command (0x85) should be acted upon immediately, never be buffered

In my opinion it should be buffered AND acted upon immediately (sounds controversial but it;s possible) , it should be cleared once the head stopped moving. By keeping the command in the buffer we have "chronology" and we know which commands to clear and which not.  Maybe I have to draw some flowchart to explain better what I mean

terjeio, 2017-03-09:
@luben111 : I believe your suggested solution will only complicate the issue at hand, remember the cancel command is acted upon when the processor is executing code in interrupt mode. This guarantees the chronology of received characters to always be 100% in sync with what is sent, flushing the input buffers while in the interrupt handler cannot possibly affect whatever arrives _after_ the cancel command. If for some reason you want to preserve any commands received _before_ the cancel it will surely fail, but IMO while jogging nothing but jogging commands (including cancels) should be sent.

NB - "input buffers" are the serial ringbuffer and the line buffer in protocol.c, they are safe to flush while in interrupt mode - and both has to be flushed together to avoid parsing errors. The correct way to flush the line buffer is by entering a (reset) character into the serial ring buffer, this ensures it happens in a clean way. Do not attempt to to reset char_counter directly from the interrupt handler.

So, if the code fails when a stream of only jogging commands are received I am of the firm opinion that the remaining problem is related to state handling in the rest of the pipeline. Perhaps adding a flag to the state machine could help, but then you need to handle that as well. I have found and patched two places where it fails (from my limited understanding of how the code should work), are there more? And if so where? The comment from @chamnit above suggest there may well be.

Anyway, I will be happy if you find a solution and/or prove me wrong. I can test any code changes you make if you want me to help out, but I am afraid I cannot suggest any more before I am able to reproduce the problem anew.

terjeio, 2017-03-09:
@luben111 : can't sleep but not due to this...

There is a time window from when the line terminator (for a jog command) is detected until sys.state is set to STATE_JOG in jog.c - if a cancel command arrives in this window it will not be able to stop it (and it has to). Among the factors making this more or less likely are baud rate, processor speed/load and keyboard debouncing/processing times. A flag could be used to stop it but the tricky part could be how to clear such a flag if there is no jog command is this part of the pipeline when it is set. On detecting the next line terminator? Somehow keeping track of the jog command until it can be safely decelerated by EXEC_MOTION_CANCEL?

Are you testing with a real keyboard or by simulation?
If by simulation do you insert a reasonable minimum delay between sending a jog command and a cancel?
Have you tested with lower baud rates? Lower processor load?

Perhaps I am just ranting now, it could well be that the scenario above will never occur in a real life situation.

luben111, 2017-03-09:
@terjeio 
I have to agree that what I proposed is maybe not efficient for implementation in existing code and requires deep changes in the whole structure (which is highly undesirable).

It makes sense that some commands may slip through serial_read() when processing CRLF - this possibility should be investigated thoroughly.

I'm testing the code on real CNC machine with Arduino UNO and PC software under LabView, 115200 bps. I use keyboard arrows for controlling the jogging, also can monitor what I sent to the GRBL.

If the problem persists I could modify the LabView code and export it as separate EXE file so everybody can reproduce it, the only one inconvenience is that you need to install LabView runtime library on your computer (free) in order to function correctly.

At the moment the jog problem occurs rarely which could be quite dangerous because you're not anymore focused and alert that something bad may happen. If you move your tool against the stock and jog cancellation is not working properly you may see how your tool crashes.

luben111, 2017-03-09:
@chamnit 
I agree - your proposed jogging control algorithm should not have issues with the existing GRBL. As I see fixing the jog cancellations may require deep changes so probably it's worth to try your approach.

luben111, 2017-03-10:
@chamnit 
There is one possible patch for the problem when using my algorithm for jogging (sending single jog commands targeting the end of the working area) which should work with the original unmodified GRBL v1.1. The idea is quite simple:

1. Send jog command to the point which is the end of the working area (if moving simultaneously on 2 or 3 axis - send a jog command to the end points of all active axis). This works pretty well and also doesn't require any communications with GRBL during the move, the head will stop automatically on the end of the working area.

2. If the user wants to stop the jog movement and we have the worst case - from two axis movement to no movement we just need to keep for 0.2s a history of the previous state and if case detected we send with delay an additional jog cancellation command. In short - if the control program detects the case when the jog cancellation will be ignored it should send with delay an additional jog cancellation.

chamnit, 2017-03-10:
@luben111 : There is an old trick that I forgot about to know when something is in sync. If you send a `G4P0` command, this will return an `ok` when Grbl is IDLE. So, if you are streaming a lot of jog commands and issue a jog cancel, just send a `G4P0` and wait for its `ok` to return before resuming more jog commands. The latency of this is minimal. Much faster than 0.2s.

luben111, 2017-03-10:
@chamnit  Thanks - it's very useful !

chamnit, 2017-03-11:
Closing this issue as it seems to be resolved. @luben111 feel free to open a new issue if you run into problems with the proposed solution.

closed 2017-03-11 by chamnit

arkypita, 2019-04-29:
Hi all, same problem here.
Jog continue even after sending Jog Cancel immediate command.

I send 

Any hint?

https://youtu.be/34LQxzKqU3w

zeevy, 2019-04-29:
I think this is what jog cancel command will do
1. Clears the planner buffer, tries to stop the machine.
2. Jog commands which are in rx buffer at this time will be added to planner buffer. 
3. And jog continues. 

I also observed no matter how many jog commands are there in the rx buffer after jog cancel, only one gets through planner buffer

zeevy, 2019-04-29:
My work arround is, dont fill the planner buffer completly. If it is filled then jog commands will get queued in rx buffer which can not be cleared with jog cancel.

In my app i do like

if(PlannerBuffer > 5) then only send jog command. 
This will make sure my jog commands wont sit in rx buffer.

arkypita, 2019-04-29:
This is what I am doing with button:

1. On mouse down I set a variable called "jog_direction" with required jog direction, and I send a first jog command.
2. In the thread that read the com port, when I receive an "ok" i check if jog_direction is not null, in that case I put a new jog command (so I put a new jog command every time the previous command receive its own "ok" reply).
3. On mouse up I set jog_direction to null (so any ok does not longer send any new jog command) and I send 0x85.

arkypita, 2019-04-29:
> My work arround is, dont fill the planner buffer completly. If it is filled then jog commands will get queued in rx buffer which can not be cleared with jog cancel.

I am not sure to have real time and affordable knowledge of planner buffer.
I mean, they are reported by "Bf:15,128" in realtime status report, but look like that they could be disabled https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#10---status-report-mask

zeevy, 2019-04-29:
Ya they can be disabled, in my app i check for that and re enables it.

You can use G4P0 to sync the buffer after each jog command, this will also solves the problem.

[10 more replies]