gnea/grbl #96: Limit Switch Wiring

tklus, 2017-01-18:
Good Morning all,
Would it be possible for someone to add a page to the Wiki that explains the correct way to wire NC limit switches. maybe with a schematic drawn up. I have been confused about this for some time and there is limited info out there that I have seen that shows how to properly connect NC limit switches to GRBL. also it would be nice to know what resistors are required as well. and what GRBL settings need to change.

I had limit switches set up NO and had nothing but problems with false triggers. I would like to get limit switches working in the NC setup.

Thanks!
Tim

chamnit, 2017-01-18:
@tklus : I'd like to see this as well, but I'm not an electrical engineer and don't want to give bad advice.

f0m3, 2017-01-18:
Well. i just connected my switches between the pin 9 ,10,12 and ground. No resistor needed because off the internal pullup. Maybe the switch is false triggered because of motor wire induction. I soldered in some small capacitor to handle this.

tklus, 2017-01-18:
and where did you install the small cap? schematic?

f0m3, 2017-01-18:
I think i did it somehow like this (i cannot go to my nc right now to check it ): 

> All you need to do for 100% perfect limit switch operation with Zero false triggers is: 3 small capacitors.
For each capacitor, connect the negative leg to the ground rail, and the positive to one of the limit sense pins on the controller. On an arduino running GRBL these are pins 9,10, &11.
For Shapeokos I have found that a .47uf is perfect in all instances.

f0m3, 2017-01-18:
BTW: some people add the capacitor close to the switch, which is useful to debounce the switch. I would prefer to have it close to the arduino to reduce noise AND to debounce. Whenever the switching wires are running close to the motor wires they will pick up some induction and this might be large enough to trigger the limit pin.

tklus, 2017-01-18:
something like this?
[image]

luben111, 2017-01-18:
The best solution is to add opto couplers (like TLP185 http://uk.farnell.com/toshiba/tlp185-y-se/optocoupler-phototrans-3-75kv/dp/2524262):
- the processor pin is not anymore connected galvanically to the end switches (no ESD noises can affect the processor)
- the opto couplers suppress high frequency noises 
Here is the schematic of my Arduino Uno shield.
[image]
[image]
[image]

langwadt, 2017-01-18:
capacitor close to the MCU pin, series resistor to the switch, it offers a bit of protection against ESD and makes a lowpass with capacitor.  A pullup sized so that the switch current at least a few mA, the low impendance is less senstitive to interference and most switches need a minium current to be reliable

gerritv, 2017-01-18:
You might want to read this about debouncing: http://www.labbookpages.co.uk/electronics/debounce.html
Also:

tklus, 2017-01-18:
@luben111, that is a nice shield. are they available for sale?

luben111, 2017-01-19:
@tklus, I could organize some small production of these modules. Let me know do you have interest, I'll calculate what I could achieve as price for assembled module and for a kit. 

This is not the last version of the board. The new version has hardware ENABLE protection - after power up the Enable signal stays non active for 1.5s (independently from Arduino board) to avoid spontaneous motor movement. Sometimes after power up you can hear the machine with GRBL to make short noise like "grrr" which can burn your drivers (caused by the initial floating pins of the controller after power up) - after adding hardware to hold ENABLE in non active state for 1.5s thighs are OK.

luben111, 2017-01-19:
BTW, some thoughts about capacitors in parallel to switches:

1. Adding large capacitor in parallel to the switch contacts is a bad idea - in long term you'll experience malfunction of the switches. The reason is that when the switch is opened the capacitor charges to power supply (5V) and when the switch closes large currents pass through the contacts (peak current could exceed 10-20A for couple of uS). Usually the small switches are rated for lower currents and in combination with moisture the life of the switch may be strongly reduced. If adding capacitors in parallel to switches it should be 10-100nF ceramic, X7R grade.

2. The electrolitic capacitor has large internal induction so all high frequency noises could not be suppressed by electrolitic capacitor.  You can notice that in all power supplies parallel to electrolitic capacitors they put some ceramic ones for suppressing the high frequency noises. So adding electrolitic capacitor on digital inputs has no suppressing effect on high frequency noises.

3. Microcontroller  pins doesn't have Schmitt trigger inputs and by adding large capacitors the voltage level will move slowly. When the voltage is passing slowly through the 0-1 threshold  the microcontroller current may increase significantly and some internal high frequency oscillations inside the silicon may occur. This could result in hanging the controller and will make the design more vulnerable to noises.

parnz, 2017-01-19:
@luben111 , maybe add a resister in series to limit the hi current when a cap is discharge?

usbcnc, 2017-01-19:
Old dot matrix has limit switch parallel with 103 small capacitor. Having large capacitor is not good to reduce noise.

luben111, 2017-01-19:
@parnz - adding serial resistor will increase the life of the switch and reduce the noises but will make much worse the problem with slow moving voltage levels on digital pins without Schmitt trigger. When the voltage changes slowly while passing the 0-1 threshold the inpur works as good analog amplifier plus the current consumption of controller jumps.

Also not on last place you'll get delay in switches reaction, for high speeds it might be a problem.

I would not recommend adding resistors on large electrolytic capacitors.

electrokean, 2017-01-19:
@luben111 actually the ATmega328 does have schmitt trigger on all digital inputs, but I agree that the capacitors should definitely not be electrolytic, but ideally ceramic with a value in the order of 100nF.
Anything significantly larger being required means you need to be implementing other noise reduction techniques, such as lower value pullup resistors, cable ferrites, shielded cables, star grounding, and even optoisolation.
BTW, your shield looks really nice :)

luben111, 2017-01-19:
@electrokean - I'm not sure that Atmel I/O have Schmitt triggers (except for TWI CLK and DATA)
http://www.avrfreaks.net/forum/inputs-gpios-avr32-family-have-schmitt-trigger-feature

I used to work many years in Atmel developing their capacitive sensors and we relied on the absence of Schmitt trigger on the inputs to get the QTouch technology running. 

In the data sheet you can see the gap between low and high input levels - this is not the Schmitt hysteresis but exactly the dangerous area where the input works as analog amplifier. You need to see explicitly Schmitt trigger features listed for the pin.

There in one more reason to believe there is no Schmitt trigger on inputs - Atmel silicons are very fast (20MHz) and Schmitt trigger design is not compatible with high speed architecture because of different delays on falling/rising edges

electrokean, 2017-01-19:
@luben111 OK, but that link is related to AVR32 which is a different class of mcu.
I'll admit the Atmel data is a bit unclear, and I wasn't sure a while back, but take a look at the current version of the ATmega328P complete datasheet (as used by Arduino Uno for a typical grbl usage example)
http://www.atmel.com/Images/Atmel-42735-8-bit-AVR-Microcontroller-ATmega328-328P_Datasheet.pdf
See section 18.2 - Figure 18-2 shows a schmitt trigger buffer, and it is described in the text 18.2.5 regarding sleep mode.  Same in Figure 18-5.  There is a "bow tie" transmission gate shown before the schmitt trigger buffer, but that is effectively an analog switch.
Finally Figures 33-25 thru 33-27 show the I/O pin thresholds and hysteresis.
The datasheet specifically mentions the "Hysteresis of Schmitt Trigger Inputs" in Table 32-10 related to the TWI characteristics, but all the above still seems to indicate schmitt trigger inputs on all digital inputs to me.
If you can show something to the contrary I'd be happy to be corrected.

luben111, 2017-01-19:
@electrokean I have to agree with you that Atmega328 has Schmitt triggers inside - Figure 33-27 is related to the Schmitt characteristics (the hysteresis).

Lemonhawk, 2017-01-20:
Wasn't the original question about connecting up the limit switches using the NC pole in order to eliminate all the noise and adding caps and resistor?  Also seems that you get a little more protection against broken wires.  So where is the discussion about connection of limit switches using NC technique and what's the set up in GRBL.

f0m3, 2017-01-20:
chamnit created a caption in the wiki for limit switch wiring. Maybe some of the experts here could help to make a (or some options) description to make a good limit switch wiring.
[https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches](https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches)

tklus, 2017-01-20:
@lemonhawk, thanks for getting this back on subject. 

As I understand it, you wouldn't need capacitors and resistors in a normally closed system. And grbl has settings to run in a normally closed system.

I would like to know how this should be wired and what settings should be changed in grbl.

Thanks!
Tim

tbfleming, 2017-01-20:
NO vs NC doesn't reduce noise, it just changes what the noise causes. With NO, noise sometimes makes the switch look like it's pushed with it's not. With NC, noise sometimes makes the switch look like it's not pushed with it is.

vMeph, 2017-01-20:
i use to have alot false triggers, I dont know how you guys have wired but i have my NC switches with the internal pullup and a external pullup with 1k resistor conected to the arduino IOREF pin and a capacitor betwen the arduino swuitchitcitch pin and GND and never had false triggers again

vMeph, 2017-01-20:
But like @tklus asked would be nice to have this type off information on wiki page with the correct schematics for NC switches.

tklus, 2017-01-20:
@vMeph, is yours hooked up like the schematic I posted above? although I don't have the resistors drawn in.

vMeph, 2017-01-20:
@tklus your schematic shows wired to NO Switch

i have set all machine with shielded cables and i have wired my switches has NC like this,  and on grbl settings changed like $5=1, never add problems again with false triggers

[image]

One off the reasons i changed to NC switches was cause the noise issues and also cause it seems to me that NC switches are a better choise, cause lets supose a switch goes bad if they wired in NC the machine will stop, if is with NO machine will not stop

i would agree with what @chamnit  saied " but I'm not an electrical engineer and don't want to give bad advice"

Would be great to have  someone that is a  electrical engineer and undestands what would be the better choise for plug in NC switches and provide information with clear schematics on wiki page, cause it seems that there is  alot debates on this subject and some says is like this and other says is like that, and becomes a litle unclear in what to do

tklus, 2017-01-20:
@vMeph just for clarification and learning, see the attached markup of your pic
[image]

Lemonhawk, 2017-01-20:
VMeph, I don't think you need the resistor or cap in your diagram.  The limit switch ports are set to use the internal pullup and in normal operation the pin is held to ground, thus preventing noise triggering the limit actuation.  If  all we need is to set $5=1 then we're done.  I think we're all looking for agreement on this.  I would really be weary of putting a 47uf cap on one of the Arduino pins - I think one of the above diagrams showed 0.47 uf, but typical noise caps would be 0.01 or 0.001 to kill HF noise.  With an NC I think the caps are worthless.

tklus, 2017-01-20:
@lemonhawk, that is what I was hoping for as well.

Lemonhawk, 2017-01-20:
Yes the diagram by vMeph is for 2 limit switches on a single axis.  If you only have a single switch the lead from the left switch (com) would go to GND.

Lemonhawk, 2017-01-20:
In all this confusion, it's the $5=1 that would be nice to have some verification that it does flip the limit switch sensing from low to high.

vMeph, 2017-01-21:
i had done before with out the external resistor and i add problems still, it looks like the internal pull up is to high and by adding the external resistor it will be in paralel with the internal pullup and the resistence will be lower, when i had only the internal pullup my machine would trigger arround 15 minutes while doing a job, sense i change things like the diagram above my machine can go for hours and never had a false trigger again

langwadt, 2017-01-21:
the internal pull-up is definitely too high, it is spec'd at some like 20-50K that is only a few hundred uA at 5V, afaict even microswitches with gold contacts spec minimum currents of +1mA

Lemonhawk, 2017-01-21:
I can see the use a 1K pullup, but it's the 47 mf that seems excessive.

vMeph, 2017-01-21:
i have updated above post sorry didnt see the new posts 
maybe the 47mf a litle exessive, mabe can go lower,but sense it worked for me i never had it change

Lemonhawk, 2017-01-21:
Seems like it would need to be really nasty noise to pull gnd up to something that would trigger.  Any idea where the noise comes from?  It would be best to try eliminating it rather than putting bandaids all over.

vMeph, 2017-01-21:
no idea where the noise was coming from, but it looks to me that most of this noise issues seems to be related with those that use cnc shield, maybe cause all circuit is packed all together in one small board and there is alot frequecies all close to eatch other and will cause triggers sense 5v is a small signal.
but couldnt agree more that putting bandaids all overs is not the way to go about it, thats why would be nice to have all to agree in how to proper conect NC and NO switches and put into wiki pages

Lemonhawk, 2017-01-21:
Makes  me wonder about using shielded wire.  Seems like the cap would  put a real load on the limit switch ports.

jahnj0584, 2017-01-21:
Would it be possible to tune the limit switches to filter out any noise,
and set the level at which a trigger is accepted, sort of like voice
activated microphones on Skype

electrokean, 2017-01-21:
I'm an electrical and software engineer (note "little e" engineer as I never completed my degrees, but I do have nearly 30 year industry experience)

Noise filtering will generally be required for both NO and NC for correct operation and homing.  The schematic diagram by vMeph is correct, but the capacitor value is way too large.  A ceramic capacitor with a value of 100nF is going to be much better at handling HF noise, and too large a value may affect correct homing pull off.  A 47uF capacitor is generally only available in electrolytic form, and those are not good at high frequency due to high ESR (even low ESR ones have a much higher ESR than a ceramic).

As discussed above between luben111 and myself, the ATmega in an Arduino does include Schmitt trigger buffers on its inputs, so they have hysteresis and can handle slowly rising or falling inputs reasonably well (slow is a relative term, but I'm talking anything more than a few tens of microseconds).  But if you want to build something that is "future proofed" for upcoming ARM controllers, they may not have native Schmitt trigger inputs and you'd do well to avoid larger capacitors than absolutely necessary.

In regard to not needing the filtering for NC switches, that is incorrect.  Even though there is a connection to GND there is finite resistance in any cable or connection, and HF noise can couple into the cable to induce spikes.  You need more copper to reduce the cable resistance, but you want flexible cable on a CNC which means less copper.

Another thing that becomes important with NC switches is what is called contact wetting, which is a minimum current needed to flow through the switch to keep the closed contact areas clear of oxidisation.  This is one reason why the internal Arduino resistance may be insufficient.  Without this your switches can prematurely fail, making use of NC no better than using NO contacts, but this does also apply to an NO switch that is rarely used.  See https://en.wikipedia.org/wiki/Wetting_current

If using shielded cable (good idea) then the shield should connect to ground at only one end (normally the powered end), and the shield generally should not be used to carry any current.  This for limit switches you really want two core shielded cable (like stereo audio cable).

I also strongly recommend adding ferrite "beads", especially on your USB cable, but also on any other cables you may be concerned about.  These should be close to the ends of the cables.  For thinner wires you can put two or three passes of wire through a torroid ferrite, but for thicker multi-core cables like the USB you can use a split clamp-on ferrite.  See https://en.wikipedia.org/wiki/Ferrite_bead

The most common noise source is brushed DC motors which generate small arcs (spark gap transmitter), but noise can come from any cable carrying current that is switching on and off - i.e. your stepper or servo motors, or even from something like a lamp on the same power circuit.

Sorry about the long comment - hopefully not too many typos :)

jahnj0584, 2017-01-21:
Safety wise, a NC switch is better. IF the wires short or break, they will
give a false trigger. This is better then homing on a NO switch that has a
broken wire and will never complete the circuit, making your machine run
off the tracks or who knows what!

electrokean, 2017-01-21:
@jahnj0584 agreed in the case of a broken wire, but not in the case of a short.  Both problems would be apparent during a homing cycle, but both could also happen to moving wiring during operation.  As I mentioned above even NC switches can have problems if not properly used, so it isn't a 100% clear argument.

Lemonhawk, 2017-01-21:
It seems to me that an antenna (the NO connection) is much better at collecting noise than a circuit to ground (the NC connection).

tbfleming, 2017-01-21:
NC is an antenna when the switch is pressed.

luben111, 2017-01-21:
@Lemonhawk  - about collecting noise, if the switch is closed the wire works as coil (loop) so any change in the magnetic field around will result in induced voltage. For "poor man" CNC connecting directly the switches to the Arduino board is acceptable but not for professional usage. Consider buffering the home switches in some way (transistor, optocoupler, some TTL, CMOS buffers, etc.), when the switch wires are connected galvanically to the controller pins any ESD and high voltage spikes can make the program counter to jump to random position or cause spontaneous change of RAM/registers.

Opto couplers are not expensive, fast enough, they are very reliable, galvanically isolating the input from output, they are simply the ideal buffering for all  inputs for CNC controllers. I could continue the list of the benefits for using the opto couplers.

vMeph, 2017-01-21:
@luben111 could you place a schematic in how you would make a conection by using optocoupler?

electrokean, 2017-01-21:
@vMeph see the schematic that @luben111 included in his comment from a few days above
https://github.com/gnea/grbl/issues/96#issuecomment-273569806

vMeph, 2017-01-21:
Thanks sorry didnt pay atention

tklus, 2017-01-21:
@luben111, I like your shield but I am using a gshield for my stepper drivers. Would it be possible to make a small circuit board that has a ground, limit pins in and limit pins out with octocoupler built in? That way it will be easy for anyone to connect limit switches with octo couplers. 

I bet there would be some interest if we ran a small batch of boards. I am not an e engineer so I wouldn't know where to begin. If someone would make a board design that would work, I could probably have the boards made and sell them at cost. Or, I have seen designs that can be posted to prototype houses and you can order the boards in small batches. Then we could install components our selves.

Thanks!
Tim

Radiu, 2017-01-21:
interesting discution for what i understand @vMeph  schematic looks good for the exeption that the 47mf is a litle to mutch and a 100nf would be ideal.
@luben111  option of using optocouplers woud be more ideal then vMeph schematic from what im reading, im having hard time to understand the luben111 shematic in what end the switch is plug in and where the arduino pin is conected?
whould it be possible to show a clear schematic of using only um axis with there right names in how is conect, like arduino limit pin and where is conect the switch pin, is that schematic for NC or NO switches
possible to have a schematic for both possibilities NO and NC?
sorry for my english and for my questions, my experience in reading shematics is a litle poor so if would be possible to have a more simple shematic more clear would be great

thanks

luben111, 2017-01-21:
See how to connect the opto couplers to switches in the existing schematic - the idea is not new and is used practically in all professional CNC boards. When the switch (NO) is closed current will start to flow through the emitting diode (LED)  which in turn will saturate the output transistor.  The LED can withstand large currents for short time and suppresses any ESD and noises. There is galvanic isolation of 5KV between LED and output transistor.

[image]

[image]

luben111, 2017-01-21:
It will take me a little more time to draw the names and make the picture as you requested. At the moment the schematic is only buffering the NO end switches, in order to support NC switches ideally I should add one IC (XOR functionality) and one jumper to invert the signals (to support NO and NC switches). Let me know if you need it.

luben111, 2017-01-21:
@tklus - Sure, I could quickly design one board with optocouplers for limit switches plus functionality to support NO and NC types (this feature was not included in the uploaded board schematic). It's "piece of cake" and will not take more than a day.

Do we have some target size of the board? Some mechanical limitations, requirement for input connectors? The resistors and optocouplers on the board - shall we use SMD (more difficult for soldering) or through hole type? For sure the SMD version will be more compact and elegant.

I can generate the Gerbers for production - do you have some PCB factory in mind?

tklus, 2017-01-21:
If optocouplers are available with pins, through mounted would be easier for diy. I have my cnc router running and have some board material on hand and could whip some up real quick. As far as no or nc, we should settle on what is going to be best for the machines. I also have a laser set up and can etch the pcb'

As far as connections, I prefer to solder the wires to the board but we could also do pin headers. I have those on hand too.

tklus, 2017-01-21:
Was thinking about this company, http://www.basicpcb.com/?gclid=CjwKEAiAqozEBRDJrPem0fPKtX0SJAD5sAyH2ECf87bvqaWrspJlZbW2YKXJ-O3Rru6jC-jvPDqgNhoCIxjw_wcB

But I haven't used them before. Does anyone else know of a reputable company?

vMeph, 2017-01-21:
I would say NC switches is always the best option for safety, lets say a broken wire, a wire lose, etc..it will always stop machine, if NO switch you will not know until you hit with machine

tklus, 2017-01-21:
This is the opto to order correct?
http://www.digikey.com/product-detail/en/toshiba-semiconductor-and-storage/TLP185(GR-TPR,E)/TLP185(GR-TPRE)CT-ND/3025457

Can we bend the legs a bit to mount it through hole?

I do have solder paste and can do surface mount if that would be better.

luben111, 2017-01-21:
@tklus - OK, through holes components. Notice that if we populate the components on the boards the SMD mounting is much better and cheaper compared to through holes. But if sell the kits - maybe through holes is better.

@vMeph  -  we'll supply the board with two wiring diagrams and recommend the NC as better and more reliable wiring (with the corresponding changes in GRBL settings $6).

I personally try to avoid soldering the wires - using connectors with screws is the professional approach and  is not that difficult to crimp the wire ends.  But let's make it simple and start with version with standard 100 mils headers.
Some idea what kind text to add on the PCB, some labels, trademark, etc.). If somehow we mention that this is "certified" or approved version from GRBL team it will make better impression. I can add the GRBL logo for example or something similar.

luben111, 2017-01-21:
Here is the schematic of the board - let me know what could be added or changed
[image]

[279 more replies]